Ultimaker/Cura #185: About json machine settings

luisonoff, 2015-07-26:
I found that Cura is not fault tolerant on this json files. I made a mistake on the json that was being used as default and could no longer open Cura until I found it was this and fixed it. Cura would open but then show a error message and close once I click OK (only option available).

By the way, mistake I made was setting a triangular area at the disallowed areas. This takes me to next question, can we only use rectangular areas for disallowed areas? How should we proceed to make circular bed shapes? Using many rectangular areas to aproximate a circle?

My json machine settings has UltiGcode flavour. I have seen that gcode I put inside start gcode in this same file does not apply, while end gcode does apply. Any reason for this? a bug?

BagelOrb, 2015-07-27:
1) the json file is not supposed to be edited by users, so investing time in making it more user friendly for those who hack the file is not work the effort.

2) this sounds to me like a bug. Triangular shapes should be possible! Can you post a sample of the disallowed area that crashed the front end?

3) I just updated that part last Friday! Now the default start gcode is empty, but it is prepended if it's not.

morriswinkler, 2015-07-31:
please read https://github.com/Ultimaker/Cura/issues/201 post number 4 , and make a common decision rather users are encouraged to change resource files or not

closed 2015-08-14 by nallath