Ultimaker/Cura #188: Multiple Monitor Support

Searider, 2015-07-26:
I have multiple monitors and Cura 15.06.02 will always load on the windows Primary monitor instead of the monitor that I closed the application on. This forces me to move the window each time I load. This is not consistent with other windows applications and a frustration.

Xisle, 2015-07-26:
The window location is not saved in cura.cfg, only the dimensions. I will also need multi-monitor support when I'm able to use new Cura to print ;)

nallath, 2015-07-27:
You're right. I'm looking into it.

fieldOfView, 2015-07-29:
As long as you are looking into that code, make sure not to save the maximised window dimensions, but instead save the original window dimensions, and possibly also save the window state (maximised or normal).

Currently if you maximise the Cura window and close and reopen the app, you get a "non-maximised" window at a maximised size, making the whole maximising/restoring a window broken.

nallath, 2015-07-29:
We're thinking about saving the whole of the window state and restore that (instead of the broken stuff that we do now)

nallath, 2015-09-02:
Fixed by fieldOfViews pull request

closed 2015-09-02 by nallath