Ultimaker/Cura #190: 15.06.2: Combing moves crossing Perimeters

CCS86, 2015-07-27:
Hi guys, I love Cura's ability to avoid retracts with combing moves, but noticed tonight that some of them are treading right on / crossing external perimeters. This is definitely affecting the surface quality:

[image]

nallath, 2015-07-27:
@BagelOrb This sounds like an engine issue.

BagelOrb, 2015-07-27:
Can you provide me with an stl which suffers from such problems under standard settings?

It's hard for me to see from your screen shot what is wrong here. Are the infill lines I am seeing on the top layer? Are you printing a cylinder with 5 outer walls and no infill, or a hollow cylinder?

Perhaps the gcode viewer of the old Cura is better.

Please provide with with an STL.

CCS86, 2015-07-27:
This is a part set to 3 perimeter shells, so in this section, we are seeing 6 shells. It is a hollow cylinder, so the inner and outer shell are outer surfaces.

It might be a little more clear in this picture:

[image]

CCS86, 2015-07-27:
Here is the STL file: https://dl.dropboxusercontent.com/u/60958586/Body.stl

BagelOrb, 2015-07-27:
I can see clearly now the view is closer. 

It seems the distance by which to stay within the perimeters of the current part is not configurable, while it should be.

This should be easily fixed, but because a new settings needs to be translated in all languages you should would have to wait a couple of months.

CCS86, 2015-07-28:
Thanks bagel, glad I could help. Hope we see it soon.

BagelOrb, 2015-08-12:
Looking at it again it looks like it's not crossing the boundary, but it's getting really close to the edge. This is also what I see when I look at the gcode produced by 15.06.

Instead of making it configurable, I've made it default that the combing boundary is on the second perimeter. This saves the time of making an extra offset, which would be the case when the distance would be configurable.

During combing, the nozzle now stays around the second perimeter.

closed 2015-08-12 by BagelOrb

CCS86, 2015-08-12:
Cool, thanks!