Ultimaker/Cura #191: Cura 15.06.02 does not honour Blenders scale setting

ThatJCB, 2015-07-27:
See also issue #180 

Cura version 15.06.02 does not honour the scale setting from Blender.

I create my models in mm and set the Metric scale in Blender to 0.001 to change meters to millimeters.

Any model I export to STL while the setting is at 0.001 will never be calculated by Cura 15.06.02. It remains as a tiny dot on the bed.

If I change that back to a scale of 1.0 before exporting the STL file then it works in Cura 15.06.02.

These same files worked perfectly and were the exact size in Cura 15.04.

ThatJCB, 2015-07-27:
There appears to be several problems with the scaling in 15.06.02.
Issues #142 and #162 are both also related to scale.

BagelOrb, 2015-07-27:
I was just about to refer to those ^^

BagelOrb, 2015-07-27:
Your model has been exported incorrectly in Blender. Perhaps you misunderstood the settings. perhaps you should set it to 1000 instead of 0.001?

The old Cura had a way of correcting for such export mistakes, which multiplied the dimensions of an object with powers of 10 (or 1000?) until the object had reasonable dimensions.

This has yet to be implemented in the new front end.

ThatJCB, 2015-07-27:
Thank you.

I have checked in Blender.
Scale setting: 0.001 the dimensions display the size I expect, example 3.1cm
Scale setting: 1000.0 the dimensions are now incorrect, same example now shows 31km

I agree it is likely that the Blender export to STL is where the conversion should be done and is perhaps not being done. My guess is that metric Blender files always output in meters instead of millimeters.

Workround: Manually set the scale to 1.0 before exporting
Potential Fix: Code a Blender Python script to fix the STL export!

Python does not suit my coding style. Too haphazard for my liking so I'm not coding that fix any time soon. I'll see if I can manage to report an enhancement request on the Blender forum.

I look forward to your auto scaling being added back however being missing it has highlighted a problem with all of my exported STL files so probably better if I fix them at the source.

Thanks again for the prompt reply.

awhiemstra, 2015-07-27:
Basically, we only scale down, since that is easy to detect. As there is no way to actually determine the scaling setting as used by blender, anything else will just end up with tricky heuristics about "what is a reasonable object size". And even the simple "scale down when bigger than build volume" heuristic is invalid as proven by other issues.

In your case, since you are modelling at 1mm scale, why do you even provide a scaling factor? It seems to me 1:1 scaling should be fine, since 1mm in Blender would equal 1mm in Cura.

ThatJCB, 2015-07-27:
Sorry, Blender only changes the cosmetic view for measurements. It still works in meters even though it displays the lengths in other metric units!

I have been experimenting in Blender and reading their developer forum.  The problem has been reported to them but they stubbornly refuse to make a change because according to them the units for STL files are not documented!

I have put in a polite feature request.

I have found some settings that get it right but to me the settings are counter intuitive.
[image]

It you use the tool box Export button those settings are not honoured so the model comes out the wrong size!

I like Blender because it is powerful and free but I despair at some of it's user interface choices!

Now I know, I'll use those settings.

'awhiemstra', I agree about your choices with auto-scaling. Scaling down is a good idea, as long as the scale factor is shown. Scaling up just confuses and hides errors in the source files.

Thank you.

BagelOrb, 2015-07-27:
@awhiemstra I don't see how a heuristic would be tricky. If (any dimension of) an object is bigger than a meter, scale down by powers of 10 until the objects fits into the volume. If (each dimensions of) an object is smaller than a millimeter, scale up until the object is lrger than a centimeter.

The problem is that an STL has no units attached. It is not fair to say that an STL with points specified as meters (so 1000 times bigger than what we expect them to be) is incorrectly exported. Since there is no way to objectively deduce the unit, we need some heuristic.

ThatJCB, 2015-07-28:
This thread #191 can be closed as it's resolution is duplicated in post #180 

All Cura 15.06 needs is a message when models are too small for it to process. See post #180.

BagelOrb, 2015-07-28:
https://github.com/Ultimaker/Cura/issues/180 is on the engine crashing for too small objects, this issue is on auto scaling too small objects.

https://github.com/Ultimaker/Cura/issues/142 is also related, it's on bugs relating to auto-scaling to just fit within the print volume.

nallath, 2015-08-24:
Large models are now sliced correctly and auto-scaling small objects is not something we want.

BagelOrb, 2015-08-24:
Why is auto-scaling small objects not something we want?

When an object is modelled in mm's but exported in meters the resulting stl will be a factor 1000 too small. We will register it as having sizes smaller than 220 micron, which is smaller than the nozzle size.
Note that this is not the fault of the modeller or the program in which it was modelled.

We should auto-scale by powers of 10, we should never scale to fit.

BagelOrb, 2015-08-24:
Such a small object is really hard to select.
An object intended to be the same size as the buildplatform becomes barely selectable and so it is really difficult to adjust the size.
[image]

closed 2016-12-26 by thopiekar