Ultimaker/Cura #192: Scale tool doesn't rotate

BagelOrb, 2015-07-27:
Steps to reproduce:
- Rotate an object 90 degrees (around the X axis for example)
- Goto scale, turn off Uniform scaling
- Scale in Z only

Expected:
- Object is scaled in Z

Actual:
- Object is scaled in Y

----

Since the scale transformation is applied before the rotation transformation, the rotation should be applied to the scale tool as well. The RGB draggable axes should point in rotated directions.

------

additional feature request: post-rotation scaling tool, which does perform scaling on the actual axes.

BagelOrb, 2015-08-24:
The scale tool still isn't rotated.

Moreover it is still buggy:
- rotate 45 degrees
- open scale tool
- drag the arrow pointing up (there shouldn't be one! It should have been rotated!)
- watch weird stuff happen as the Z percentage suddenly snaps to 0%

nallath, 2015-08-24:
I only checked for 90 degree rotations. It should be correct now.

closed 2015-08-24 by nallath