Ultimaker/Cura #193: Tools for rotating/scaling do not appear when model is selected

remy-spencer, 2015-07-27:
Build plate background does not load, and when rotate/scale is selected none of the tools appear (rgb circles for rotate, arrows for scale). Occurred on two computers running Win7.

[image]

nallath, 2015-07-28:
Could you post the log files and your computer specs?

Mikepwsm, 2015-07-28:
I'm also experiencing the same problem, Win 7 Ultimate SP1 64 bit. AMD Phenom II P920 Quad Core 1.6 gHz 8 GB Ram.

Logfile:

Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ActiveToolProxy.py", line 61, in setProperty
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\Tools\ScaleTool\ScaleTool.py", line 236, in setScaleX
    obj = Selection.getSelectedObject(0)
  File "c:\dev\uranium\UM\Scene\Selection.py", line 34, in getSelectedObject
IndexError: list index out of range
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\VisibleMessagesModel.py", line 64, in actionTriggered
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\cura\cura\CuraApplication.py", line 528, in _onMessageActionTriggered
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LocalFileStorage\LocalFileStorageDevice.py", line 29, in ejectRemovableDrive
    self._removable_drives.ejectDrive(name, self._drives[name])
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LocalFileStorage\WindowsRemovableDrives.py", line 91, in ejectDrive
    if not windll.kernel32.DeviceIoControl(handle, IOCTL_STORAGE_EJECT_MEDIA, None, None, None, None, None, None):
OSError: exception: access violation writing 0x00000000
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 155, in event
  File "c:\dev\uranium\UM\Event.py", line 149, in call
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Backend\Backend.py", line 128, in _onSocketError
AttributeError: 'UnknownProtocolError' object has no attribute 'errno'
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 10
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 9
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 8
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 7
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 6
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 5
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\LayerView\LayerView.py", line 98, in beginRendering
[... 140 lines cut ...]
IndexError: list index out of range
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ActiveToolProxy.py", line 61, in setProperty
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\Tools\ScaleTool\ScaleTool.py", line 247, in setScaleY
    obj = Selection.getSelectedObject(0)
  File "c:\dev\uranium\UM\Scene\Selection.py", line 34, in getSelectedObject
IndexError: list index out of range
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ActiveToolProxy.py", line 61, in setProperty
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\Tools\ScaleTool\ScaleTool.py", line 210, in setObjectHeight
    obj = Selection.getSelectedObject(0)
  File "c:\dev\uranium\UM\Scene\Selection.py", line 34, in getSelectedObject
IndexError: list index out of range
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ActiveToolProxy.py", line 61, in setProperty
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\Tools\ScaleTool\ScaleTool.py", line 210, in setObjectHeight
    obj = Selection.getSelectedObject(0)
  File "c:\dev\uranium\UM\Scene\Selection.py", line 34, in getSelectedObject
IndexError: list index out of range
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ActiveToolProxy.py", line 61, in setProperty
  File "C:\Program Files (x86)\Cura_15.06.02\plugins\Tools\ScaleTool\ScaleTool.py", line 247, in setScaleY
    obj = Selection.getSelectedObject(0)
  File "c:\dev\uranium\UM\Scene\Selection.py", line 34, in getSelectedObject
IndexError: list index out of range

remy-spencer, 2015-07-28:
Specs: Dell Optiplex 980 running Win 7 with a Radeon HD 4550 video card, Dell Optiplex 980 running Win 7 with a Radeon HD 3400 series video card

Looking into getting the log files as well.

gmrishel, 2015-07-29:
nallath & remy,

I am the user that first contacted remy with this issue, remy thanks for the help so far.

Nallath, where should the log files be located?  I searched Cura  the dir for *.log  - no file found. 

In Cura under help "show engine log" which I can view, but when I attempt to c&p, it un-highlights the text before I can get scrolled to the bottom.  Maybe part of the issue?

Please advise.

nickthetait, 2015-07-29:
I think it would be awesome to get a "show log file" button that opens a file browser to where the log file is located. Would save lots of headache for engineers and users. But this is a distraction to the current issue being discussed. @nallath, would you mind me creating a new issue for this?

gmrishel, 2015-07-29:
nickthetait, do you have any idea of the log file path or name?  If I find them I will try to post them.

c:\Program Files (x86)\Cura_15.06.02>    ???
or
c:\Program Files (x86)\Cura_15.06.01>   ???

Both versions are having this issue, 15.04 is working for me.

awhiemstra, 2015-07-29:
Log file locations are listed under https://github.com/Ultimaker/Cura#logging-issues

Unfortunately, there is currently an issue where the crash handler does not intercept exceptions that end up in the Qt event loop so you may also need the file created in the installation path when running as Administrator.

gmrishel, 2015-07-29:
Thank you awhiemstra!
Since I have 3 versions installed, i delete the contents and tried again.  After closing this is what I have:

2015-07-29 16:31:07,165 - INFO - Loaded plugin FileLogger
2015-07-29 16:31:07,656 - INFO - Loaded plugin LocalFileStorage
2015-07-29 16:31:07,668 - INFO - Loaded plugin LayerView
2015-07-29 16:31:07,687 - INFO - Loaded plugin MeshView
2015-07-29 16:31:07,713 - INFO - Loaded plugin WireframeView
2015-07-29 16:31:07,721 - INFO - Loaded plugin OBJReader
2015-07-29 16:31:07,726 - INFO - Loaded plugin STLReader
2015-07-29 16:31:07,736 - INFO - Loaded plugin STLWriter
2015-07-29 16:31:07,746 - INFO - Loaded plugin GCodeWriter
2015-07-29 16:31:07,761 - INFO - Loaded plugin CameraTool
2015-07-29 16:31:07,777 - INFO - Loaded plugin MirrorTool
2015-07-29 16:31:07,807 - INFO - Loaded plugin RotateTool
2015-07-29 16:31:07,825 - INFO - Loaded plugin ScaleTool
2015-07-29 16:31:07,840 - INFO - Loaded plugin SelectionTool
2015-07-29 16:31:07,858 - INFO - Loaded plugin TranslateTool
2015-07-29 16:31:07,879 - INFO - Checking for new version of cura
2015-07-29 16:31:07,879 - INFO - Loaded plugin UpdateChecker
2015-07-29 16:31:08,130 - INFO - Loaded plugin USBPrinting
2015-07-29 16:31:08,133 - INFO - Loaded plugin CuraEngineBackend
2015-07-29 16:31:10,186 - DEBUG - Loaded theme C:/Program Files (x86)/Cura_15.06.02/resources/themes/cura
2015-07-29 16:31:10,639 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,639 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,906 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,907 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,907 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,908 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,908 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,909 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,910 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,910 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,911 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,911 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,912 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,912 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,913 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:10,913 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:11,268 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:11,269 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:11,380 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:11,381 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:12,317 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.02\CuraEngine.exe
2015-07-29 16:31:14,640 - DEBUG - Backend connected on port 49674
2015-07-29 16:31:22,417 - DEBUG - Loaded a mesh with 3798 vertices
2015-07-29 16:31:23,112 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,113 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,113 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,114 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,114 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,115 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,115 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,116 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,116 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,139 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,140 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,140 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,141 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,141 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,142 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,142 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,143 - ERROR - An error occurred in inherit function for wall_line_width: division by zero
2015-07-29 16:31:23,145 - ERROR - An error occurred in inherit function for wall_line_width: division by zero

nallath, 2015-09-23:
@awhiemstra Do you know if this is fixed in the meantime?

Mikepwsm, 2015-09-23:
sorry I don't know. As poor response I've dumped cura in favour of
Simplify3d...

nallath, 2015-09-23:
Would you still be interested in checking if this works with the latest version? You can find it on http://software.ultimaker.com/Cura_closed_beta/15.10/

BagelOrb, 2016-04-26:
Fixed

closed 2016-04-26 by BagelOrb