Ultimaker/Cura #194: Bigger Font when calculating print time

FuzzySadist, 2015-07-28:
Hey, I admit it, I'm old... but I would love it if the print time at the bottom was a bigger font.  I know it uses a bigger font at the top right, but I prefer the "Days, Hours, Minutes" format at the bottom right.

[image]

nallath, 2015-07-28:
We're updating the GUI somewhat for the next version. Not 100% this is in it, but we will take a look at it. @TMHogenhout What do you think?

fieldOfView, 2015-07-29:
The text "Estimated Print-time" is overly verbose. I would change it to "Estimated", "Est." or even "", because it not only shows print-time but also the estimated amount of material. With the text a bit less verbose, the default font-size can be chosen.

See:
https://github.com/fieldOfView/Cura/commit/05d7ce51193340a52c71d1bca44822b5a63f0b44

Xisle, 2015-07-29:
Any hints (previews) of the GUI changes for printing? I heard the print dialog was going away. Maybe the area used for the printing status can also fit the print time, material length, mass and cm^3 for ease of calculating hub orders.

TMHogenhout, 2015-08-01:
Good point! The GUI is on a wire-frame level at this point. But we will look into this specific part on monday and upload some designs. Just for my own reference:
- Bigger! (default font size instead of small)
- "Estimated Print-time:" -> "Est:" | "" | "+/-" 
- Keep 'longer' languages into account like Russian
- Allow for the material length to be replaced by g | cm^3 or even for g & cm^3 to be added to it

What do you guys think of this idea?
- Make the entire sidebar draggable and the contents more fluid
- Allow for the different flavours of material display (length, g and cm^3) to be added to the view and removed using the preferences dialog

Xisle, 2015-08-01:
I like your ideas and look forward to seeing some designs.

fieldOfView, 2015-08-01:
Personally I think the whole "gcode" section is a bit lost at the bottom of the print setup. I think I miss having it in the top left. If you plan on keeping it in the sidebar, you may consider adding a Section Header, like the "Print Settings" section header on top. As a matter of fact, this gives the "Print Settings" header a reason to be there in the first place (as it is now, it just takes up valuable space, because there is no other section).

Making the sidebar with settable would be nice, but it does not fix this particular problem (or make the fix easier). Making the "Estimated Print-time:" text shorter does both.

fieldOfView, 2015-08-02:
For further discussion of the UI, I wonder why - from a user's point of view - the slicer has its own progressbar, whereas other processes that take a while (like loading the layers in Layer view) have a progressbar in the messagestack area. I understand it from a technical point of view (Cura progress vs CuraEngine progress), but for a user it is two areas to keep track of mentally.

fieldOfView, 2015-08-03:
@TMHogenhout, not that I am nosy or anything, but interested and curious... Yeah, that would be me ;-)
Anything to share from the design-meeting?

TMHogenhout, 2015-08-03:
@fieldOfView yes and no. I discussed it with our UX-designer, and he was working on it, but he ran into some other problems, so he will probably finish it tomorrow and put the designs here on github. Nice to know that you are taking an interest! To be continued..

mennoberg, 2015-08-04:
[image][image]
[image]

fieldOfView, 2015-08-04:
I'm not sure if I should look at this as a wireframe only, but I must say I like the direction this is going esthetically. Kudos on that. To me this looks much better.

The Machines section takes up valuable space, and I wonder if the majority of users have multiple types of printers. Even if you have a farm of Ultimaker 2 printers, you are still not going to use that whole section. I think this is much better kept to the selection in the "Machine" menu. I can see how making the Connectivity a machine setting simplifies the results area at the bottom, but I don't think creating a divide between the save button and the type of destination is a good idea.

The Advanced print setup is a step back as shown here; we need to have as much overview of the selected settings as possible. The collapsible sections in the current version actually work for that, but there is way too much whitespace. In this new design, you have to click through each of the sections to see what options have been set. The new design is also much wider than the current design, but I must say I don't think that is necessarily a bad thing. Perhaps something inbetween.

The advanced settings proposal also contains a Save button, to save changes to a profile I presume. I think we would need support for "partial profiles" for this to be useful. IE: I want to be able to save and restore the temperatures, filament diameter and flow settings without affecting the infill and support settings, so I can for example save a set of sub-profiles for different materials without changing the "structural" settings that concern the geometry of the print. One of the applications I know that does a good job at that is Adobe Camera Raw/Bridge.

In the Print Job section, what does the button next to the estimate do? Esthetically I am not a fan of very wide buttons that just take up the full width because theres nothing else to take up space. If the Print Job section is staying at the bottom, it needs a cyan header for consistency. I don't think it is necessary to spell out each unit (time, length, mass, price); "Estimated time and material" is clear enough, I can figure out that 300g is not the length of filament. Like I said before, I wonder if this section "deserves" a progressbar of its own from the users point of view. Why does loading have a progressbar in the message-stack in the viewport area, while saving has a progressbar in the Print Job section of the sidebar?

I would much prefer to have the whole "Print Job" section at the top left, like it was with legacy-cura. Then you have a "input/output" section at the top left, a "model-manipulation" section at the bottom left, and the right is reserved for only "print settings".

mennoberg, 2015-08-04:
Thanks Aldo for your valuable detailed feedback!

Without diving to much into all your feedback there some things I like comment on/
- The temp, filament diameter and flow settings exist in the printer, because these settings concern only the physical um2 machine.  We will also keep them there so cura will only been used for structural model settings, though I know there are exception for older machines.

- Have to agree with you that the machine section takes up some valuable space, though in some cases like above it has to be clear that changes or information in this area only concerns to machine and not the print structures. Will look into alternative place and how that will feel in the interface.

- I like your points on the divided section between your destination and save button, will look in to that as well how this maybe could be more clear.

Concerning the Input/output section point I have to disagree with you in a matter of flow it is i.m.o. not really a "predictable" position. Concerning workflow It fails in the begin-end feeling you should have  within the flow of making a print job.

fieldOfView, 2015-08-04:
Not everyone prints with a UM2 (UMO, other brands of printers), and even some people with a UM2 turn off the ultigcode feature for better USB printing support. Also sub-profiles are useful for more than material properties: eg optimised settings for different nozzle diameters (this may include not only linewidths, but also print speed).

The machine section already has a good place: the Machine menu. I have yet to be convinced of a reason why this fringe-functionality should be duplicated outside the menu.

Ofcourse you don't have to agree with anything I say ;-). My advice has no basis more reliable than my own meandering experience

fieldOfView, 2015-08-04:
For reference, here is my current hacked Cura:
[image]

* Much smaller margins, so there is lots to see at the same time
* Machine selection is gone
* Two (distinctly colored) headers for Print Setup and Print Job

The only functionality I miss here is a Simple/Advanced switch. I like your switch in the "Print Setup" header.

mennoberg, 2015-08-05:
Collected a lot of valuable UI feedback last days.

- Simple users would like a bit more settings
- People like to name their job
- Visually functions should look and be unique;
Estimation field looks like a text field 
The print setup looks like you can click on it while you can't
Progress bar in estimation field
- To much white space etc.

[image]
[image]
[image]

fieldOfView, 2015-08-05:
I like this! It's got a bit of that "buttons in all the 4 corners" vibe (remember Fitt's law:https://en.wikipedia.org/wiki/Fitts%27s_law), and the message-stack looks a bit misaligned (https://github.com/Ultimaker/Cura/pull/205), but that's about all I can come up with for now. Sidepanel looks great.

fieldOfView, 2015-08-05:
Perhaps have the bottom-left manipulation "toolbar" top-center? That way you have a single toolbar, divided in three parts (open, manipulate, viewmode). It would make this an inverted L interface instead of the buttons-in-each-corner syndrome.

TMHogenhout, 2015-08-06:
+1 I really like this inverted L pattern idea, because it is a pattern used very often, so it could make the interface very intuitive especially for starting users.

mennoberg, 2015-08-06:
Update: 
- Moved the tools and functions to top bar
- Color changes in active accordion menu
- Removed icons accordeon menu
- Expanded space for functions in accordion a little bit
- Added a dropdown function in accordion menu (settings under setting)
- Changed the 'Save location dropdown' to "textual dropdown'
- Changed all dropdown arrows

Please @TMHogenhout @fieldOfView  let me know what you think!
[image]
[image]
[image]

fieldOfView, 2015-08-06:
Nice alignment of the top buttons with the top of the sidebar! We're not seeing the MessageStack, but I assume it is meant to be at the bottom (no margin) centered horizontally?

We are getting into "nitpick territory" with the feedback now, looking good overall.

The setting of the advanced switch is not immediately obvious when it is turned off; it could be interpreted as a header. The switch you had in your first sketch is clearer, but may  simply take up too much space.

I liked the version of the settings list without the cyan headers and with icons better. In this version, the left margin of the settings is a bit inexplicable (the highlight-line lined up with the icons in the previous version; without the icons, it could be a full em more to the left). 

Semantics: "Local to Disk". Wouldn't "to Disk" be just as well? ("to Disk", "to SD", "to Printer" (usb printing), "to 3Doodler", "to OctoPrint"). The word "Local" doesn't add that much.

The infill icons should probably reflect the type of infill Cura uses by default, ie diagonal lines instead of honeycomb. 

Scale-Rotate-Mirror makes more sense to me than Mirror-Scale-Rotate

TMHogenhout, 2015-08-07:
I also love where this is going. It feels very intuitive and simple. But it also has a grown-up/professional overall look. Just some minor things from the nitpick territory as well. 
- Please put a screen shot of the actual build plate in the background before deciding on the final design. We can change the colours of the build plate, but we have tried several different build plates and this approach remains the best. So in large we should adapt the UI-design to the build plate not the other way around. 
- The simple/advanced toggle is not really clear. It was a lot better when there was an actual toggle. 
- I am afraid that the blue background of the print setup causes the same problem we have now; that users think that it is clickable. 
- I liked the previous advanced mode with the icons better as well, because the hierarchy was more clear. In that design you can also leave the print setup with a blue background, because the colour is no longer and indication that something is 'active'.
- I love the approach with the icons for the simple mode, but it has to correspond with the actual result. 
- Maybe there is a way that we can make it more clear that people can type their own filename before the .gcode part.

fieldOfView, 2015-08-07:
Here's my go at combining the sketches and our latest feedback:
[image]

Could the Machine dropdown only be there when there is more than 1 machine setup?

fieldOfView, 2015-08-07:
One thing missing in the recent designs is a scrollbar. Even if you can scroll by dragging anywhere, a scrollbar serves the purpose of showing how much more there is, and where you are. Then again, when you show something resembling a scrollbar, people want to use that to scroll, eg:
https://github.com/Ultimaker/Cura/issues/227

TMHogenhout, 2015-08-10:
Hey Aldo, I really like your design! You're taking a very realistic approach. Thank you for all your time and effort! I especially love the save-button area on the bottom-right. The line around the <name>.gcode makes it so much more clear that you can type there. As far is I'm concerned this would be ripe for implementation. 

@mennoberg I am only missing two things in the design:
- Where does the progress-bar (or other progress indicator thing) go? 
- What to do with the sidebar? It agree that it really can't stay in the same form it is now.

fieldOfView, 2015-08-10:
@TMHogenhout It is hardly "my design", I just cobbled together different iterations of Menno's work into what I think is the (almost) golden combination.

> Where does the progress-bar (or other progress indicator thing) go
I vote for MessageStack! One progressbar to rule them all.

> What to do with the sidebar?
Scrollbar, right?

TMHogenhout, 2015-08-10:
True. 
Good point. 
Typed a bit too fast. I mean scrollbar.

mennoberg, 2015-08-13:
Thanks for all the feedback
Update:

- Changed the Simple advanced toggle
- Message stack: Added notifications
- Message stack: Altered Slicing notification
- Simple modus, extended Icon selection to all functions
- Simple modus, changed from 4 to 3 selection. 
- Tools: changed order of tools to Scale-Rotate-Mirror
- 3D Build plate: Inserted Realistic version (temporarily but for upcoming release)
- 3D Build plate: Start orientation must be the same at start up program
- 3D Build plate: Simplified Content focused 3D build plate, extended with logo and build plate clips (future release)

- Advanced modus: Stripped line in sub menu
- Advanced modus: Accordion menu arrow orientation 
- Advanced modus: Accordion Menu bar colours & icons
- Advanced modus: Added scroll bar

- Save & result menu: Improved Name job field
- Save & result menu: Save button Background during slicing grey instead of overlay
- Save & result menu: Icons before stats

Let me know if you have any comments on details or mayor functions :)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

awhiemstra, 2015-08-13:
I really like the designs, though I have a few small things to nitpick:

- The "Load File" button is not the same size as the other top buttons, which looks really strange in my opinion.
- Putting all the tool controls in the middle top means obscuring a lot of build plate in that area. I am not sure how much people like that. Addtionally, remember that there will probably be at least six different buttons there and that the additional controls may be larger due to language.
- Since the tool buttons are now together it makes it even harder to see that they are clickable. This is probably not something we can solve this iteration, but it is something we need to consider.
- Doing the dashed line along the side of the settings is going to be really tricky to implement technically, so do not count on it looking that way in the final version.
- Shouldn't we hide the 'general print profile" if we have a "quality" just below that?

Ghostkeeper, 2015-08-14:
Really like those icons there, by the way. Super intuitive.

fieldOfView, 2015-08-14:
> The "Load File" button

I think I would prefer an icon button (though one that is a bit less cryptic than the current one). On the other hand I like how all disk operations are black buttons.

> Putting all the tool controls in the middle top means obscuring a lot of build plate

This shows especially with the rotate dialog showing (4th image). Not sure how to handle this, except by letting the use drag that dialog to a more convenient location. Other than the dialogs, I don't see how the buttons take away too much of the 3d area.

> remember that there will probably be at least six different buttons

6?

> Doing the dashed line along the side of the settings

I don't think they are necessary. I made a version with slightly different alignment:
[image]
* The header-icons align with the other text in the sidebar, and the right-margin of the icons is a bit smaller. The treeview is aligned to this.
* I made the control-column of the sidebar a bit wider. We're never going to fit "Touching buildplate" in there, but I feel we needed a bit more space either way.
* The scrollbar is wider, because people are going to want to use that to scroll. The scrollable pane now right-aligns with other controls in the sidebar.
All in all, slightly condensed, yet more room.

> Shouldn't we hide the 'general print profile" if we have a "quality" just below that?

Or at least drop the "general" and make it a "print profile". For the mockup, the functionality would be clearer if it says "Menno's settings". Because that's the idea, right? That you can store combinations of settings?

I like how simple mode gets a couple more options (thus becoming more useful).

* The quality-options go "up", the others go "down" (or have "None" at the end).
* I would say "high" instead of "highest" (who's to say that lineheight can not be smaller on this printer?)
* Surface adhesion and support icons are a bit cryptic. 
* Should there be a support option for "touching buildplate" vs "everywhere"?

TMHogenhout, 2015-08-14:
Wow I really like where this is going. I really like the widened sidebar and the widened sidebar control as well. I would suggest two additions though: 

1. A useful and well designed tool tip that could (maybe in the future) be expanded to a tool tip that can include animations/illustrations that show the use of a setting in action. 
2. We would like to do something with the Load File button when there is no file loaded (make it more blue or something to indicate that you should start with loading a file). 

Here is just an idea. What if we would make a software start-up tour (not for the 15.10 release but later). Really cool example here: http://www.jqueryrain.com/?gY_FlAOC If we want to do something like this later on, we could already design the Load Button on start-up (when no file is loaded yet) already in such a way that it combines well with a first start-up tour.

TMHogenhout, 2015-09-22:
The font of the print-time etc will be default size (taken from your platvorm settings) as from the 15.10 release. 
[image]

closed 2015-09-22 by TMHogenhout

iakapulko, 2015-09-22:
About the intuitiveness of the interface: the file load button (and others too), imho, should have an option to choose between icon and/or signature. The model morphing buttons (scale, etc) in the lower left corner are very pretty, but option with the same buttons in the upper part in the center makes my eyes hurt, diverting attention from the workspace

fieldOfView, 2015-09-22:
@thhogenhout: BAM! back on topic ;-)

@iakapulko: it is probably better if you open a new issue for this. Possibly with a sketched proposal on how you would improve the UI from the current state (15.09.8x)

TMHogenhout, 2015-09-24:
@fieldOfView :D