Ultimaker/Cura #195: Feature Request: Auto Arrange

FuzzySadist, 2015-07-28:
It would be great when placing more then one item on the tray, if Cura could either auto arrange them, or barring that had a button to do so.  Especially nice if it chose placement based on best printing practices.

nallath, 2015-07-28:
Auto arranging is hard. It already does it somewhat, but finding the correct location of a bunch of objects -and- take best practices of printing into account is even harder. 
This feature is not on our roadmap and due to its complexity it's not very likely that we will create it. People are free to create a plugin that does this.

closed 2015-07-28 by nallath

guzforster, 2018-03-30:
Better than that would be a feature that allows users to select multiple models and align and distribute along X, Y or Z axis. And would also be nice to have a duplicate function that does the same: select one model, make an array of X objects along X, Y or Z.

Ghostkeeper, 2018-04-26:
The auto arrange has now been implemented by the way.

What do you think is the added value of arranging them along one of the axes? Like, in a line?

kickahaota, 2018-06-11:
Auto-arranging is now implemented, and it works pretty well for very small items. But the current implementation fails pretty badly on large objects.

To give a simple example: Load a calibration cube and scale it to 80x80x80 mm. A human can fit four of these onto the Ultimaker 3 build plate. But the auto-arranging algorithm in the current 3.3.1 and 3.4-beta versions of Cura can't arrange two of them -- it places the first one squarely in the center of the build area, and then finds that there's not enough room in any direction for a second one.

I certainly understand that arbitrary-object packing algorithms are hard. But it seems like Cura ought to be able to better manage simple cases involving rectangular objects. Just basic things like "try starting in a corner rather than in the center".