Ultimaker/Cura #196: "speed_layer_0" should be child of "speed_print" in fdmprinter.json?

Xisle, 2015-07-28:
I think the Bottom Layer Speed should default to the current Print Speed when not visible.?.

Xisle, 2015-07-28:
I edited the .json to make "speed_layer_0" (and its child) a child of "speed_print" and it works as expected.

I can see justification for it being either way now that I've switched it.

nallath, 2015-07-29:
Make it into a pull request, that's what they are for :)

BagelOrb, 2015-07-29:
The thing is that as thing are currently, making speed_layer_0 invisible will make its value inherited from speed_print, while we might have better defaults than that.

Xisle, 2015-07-29:
speed_layer_0 does not currently inherit from speed_print as it is not one of its children and by default speed_layer_0 is invisible and set to 15.

It seems like speed_layer_0 should either be visible by default or made a child of speed_print and left invisible so it gets inherited value.

I will gladly make a pull request if needed. I'm just looking for input first.

BagelOrb, 2016-04-26:
I don't see much need to...

closed 2016-04-26 by BagelOrb