Ultimaker/Cura #197: Keyboard shortcut for Print

Xisle, 2015-07-28:
Could you add a Print option to the File menu with associated shortcut (CMD-P or CTL-P)?

nallath, 2015-07-29:
What would the behavior be? Print if a printer is connected with USB?

Xisle, 2015-07-29:
Add "Print" to File menu but make it inactive (gray it out) when "Print to USB" is NOT available.

The Print menu item can be inactive when the print button is gray..

[image]

..and active when it is blue (or would be blue, even if "Save to SD" is currently selected)

[image]

nallath, 2015-08-20:
We're having some internal discussion as to what good behaviour would be in this regard, as pressing ctrl + p and then seeing a save to local file menu pop up is somewhat confusing.

Xisle, 2015-08-20:
ctrl-p and ctrl-s should work as expected (when printing or saving is available) regardless of the current blue button status. Why would ctrl-p show a save dialog?

The user could have "Save to USB" selected and still hit ctrl-p to print just like ctrl-s should work even when "Print to USB" is selected.

nallath, 2015-08-20:
Because we don't have an internal difference between save to file and send to printer. Both are output devices.

BagelOrb, 2016-02-04:
Ctrl-P normally (in most standard programs) pops up a window with all printer settings (color/black-white etc.) and profiles (quality etc.). It is in this popup window where you can hit the button to actually print.

The analogue in 3D prining should be that Cura starts up when you press Ctrl-P from Solidworks for example.

I think it's a bit scary to connect a real-world action to a keyboard short-cut - when you accidentally hit it and don't notice, a print is started and material might be wasted.

closed 2016-02-04 by BagelOrb