Ultimaker/Cura #199: Cura "uncaught exception"

5macksmobile, 2015-07-29:
I am running the latest versions of Cura on a machine running windows 8.1 and was working fine until today when I received this error any ideas on how to fix this?

Thanks
Mark
Ooops

Version: 15.06.02
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 124, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__
KeyError: 'general'

nallath, 2015-07-29:
This doesn't look like the latest version, as this was an issue with the first version that we release. This issue was patched in 15.06.01
You could try to remove the .cfg file in $User/AppData/Local/cura/ and try again.

ripper121, 2015-07-29:
Version: 15.06.02
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1044, in _read
configparser.DuplicateSectionError: While reading from 'C:\\Users\\STS/AppData/Local/cura\\cura.cfg' [line 20]: section 'view' already exists

ripper121, 2015-07-29:
Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1044, in _read
configparser.DuplicateSectionError: While reading from 'C:\\Users\\STS/AppData/Local/cura\\cura.cfg' [line 20]: section 'view' already exists

ripper121, 2015-07-29:
I delete my Settings under : C:\Users\User\AppData\Local\cura
Then it works.

nallath, 2015-07-29:
I believe this is an error caused because you had an old version installed before. Did you instal 15.06.00 (or a beta) by any chance?

ripper121, 2015-07-29:
No is started with the 15.06.01

nallath, 2015-07-29:
Odd. Well, it's fixed now and doesn't happen with a clean install.

closed 2015-07-29 by nallath

5macksmobile, 2015-07-29:
Thanks Jaime, that sorted it out :-)