Ultimaker/Cura #201: Supported way to add .json machine config outside of program files

dantman, 2015-07-30:
The new version of Cura shipped with no non-Ultimaker machine config and no editable machine settings. And it looks like the intent is for all machine settings to be part of a .json file either distributed for printer types or made with [the generator](https://quillford.github.io/CuraProfileMaker/). Which is a perfectly valid way forward.

The problem currently is actually using those within Cura in a way that isn't a hack. Currently it seems the recommendations are that people add a file to `C:\Program Files (x86)\Cura_15.06.*\resources\settings` on Windows and `/Applications/Cura.app/Resources/cura/resources/settings/` on OSX. Which basically means modifying the program files in a way that will likely be destroyed when the user upgrades.

I like the idea of using .json files. But moving forward Cura is going to need either one of these features:
- The ability to add custom machine config to a location like `$HOME/.config/cura/settings/` and documentation on how to.
- An option in the "Add Printer" menu that allows a user to browse and select a .json file to use as the machine config.

nallath, 2015-07-30:
New versions of Cura are installed side by side, so old resource settings will not be destroyed.

dantman, 2015-07-30:
@nallath There are still number of problems.

Firstly, on OS X you're wrong. "Installing" often means downloading and replacing `Cura.app`. Which means the entire app is replaced and the user's machine configurations are deleted. At the very least if the user renames the old Cura.app they still have to re-import all their machine configuration.

Secondly, this is a completely user hostile way for machine configuration to work. Windows users have to dig into their `Program Files` folder, which users are normally told to stay out of. On OS X they have to dig into the "Show Package Contents" view and mess with an `.app` in ways many users don't even understand exist.

Simply closing this bug is not an acceptable situation to leave users in.

nallath, 2015-07-30:
I wasn't aware on how OSX handles things. Another thing to add to my list why I dislike OS X.  I do disagree with the "users aren't supposed to dig in program files" mantra. I've never been told that I should stay out of this and you're the first user to ever mention this to me. It could be that my sampling is biassed though. 

Personally I think that is a pretty normal way of doing things. But as we will need to change things because OSX does things differently, it will change it for windows as well. 

That being said; If you find a solution not acceptable, you're free to implement the functionality yourself (Cura is open source after all!)

awhiemstra, 2015-07-30:
I actually have plans to also support loading setting definition files from the user's setting storage path. We already have support for this for plugins and doing the same for settings makes sense and is fairly simple to implement. It mostly is a case of "I just never got around to it". Interestingly enough, this issue is actually the biggest problem on Linux, since you are not really meant to change /usr/share files outside of your package manager.

morriswinkler, 2015-07-31:
Well some ideas of file locations for imported settings/machine.json files

Windows C:\Users\<user>\AppData\Roaming\Cura\settings

OSX /Users/<user>/Library/Application Support/Cura/settings

Linux /home/<user>/.cura/settings
 
tag the json file with a version tag, so you can figure from what version that json file has been, if you ever change your json format dramatically include a peace of code that is able to upgrade old setting  files

just as an idea, than you could add setting files by the add machine wizard

nallath, 2015-07-31:
I've already updated the machine settings somewhat so that machines with the same ID are grouped. This should also help with this, as you can simply change the name and keep the ID intact, showing a second choice when adding machines (eg; select um2 and then get to pick between um2 15.06.01, 15.06.02, etc)

awhiemstra, 2015-07-31:
Things will be relative to the current used storage paths, so $User/AppData/Local/cura on Windows, $User/.cura on MacOSX and $User/.local/share/cura on Linux.

nallath, 2015-08-24:
This is supported in the 15.10 release.

closed 2015-08-24 by nallath