Ultimaker/Cura #202: Cannot Run Cura

EstebanPacheco, 2015-07-30:
[image]

Hello! im habing a problem, i just dowloaded the last version Of cura, and i can install it, but, i cant run it, it crashes Always...

Yosemite 10,10,4

:(

EstebanPacheco, 2015-07-30:
I"ve got the Latest Revision (15.06.03) and still the same problem, i install it, but im unable to run it>

30/07/15 23:53:20,334 SynologySIMBLAgent[651]: Got bundle path '/Users/estebanpacheco/.Synology/Library/FinderPlugins/SynoMacIconOverlay.bundle' from config file '/Library/Application Support/Synology/SIMBL/SynologyCloudStation.config'
30/07/15 23:53:20,504 Cura[15989]: 2015-07-30 23:53:20,503 - INFO - Loaded plugin ConsoleLogger
30/07/15 23:53:20,527 Cura[15989]: 2015-07-30 23:53:20,526 - INFO - Loaded plugin FileLogger
30/07/15 23:53:20,660 Cura[15989]: 2015-07-30 23:53:20,659 - INFO - Loaded plugin LocalFileStorage
30/07/15 23:53:20,666 Cura[15989]: 2015-07-30 23:53:20,665 - INFO - Loaded plugin MeshView
30/07/15 23:53:20,669 Cura[15989]: 2015-07-30 23:53:20,668 - INFO - Loaded plugin WireframeView
30/07/15 23:53:20,674 Cura[15989]: 2015-07-30 23:53:20,673 - INFO - Loaded plugin LayerView
30/07/15 23:53:20,676 Cura[15989]: 2015-07-30 23:53:20,675 - INFO - Loaded plugin OBJReader
30/07/15 23:53:20,677 Cura[15989]: 2015-07-30 23:53:20,676 - INFO - Loaded plugin STLReader
30/07/15 23:53:20,679 Cura[15989]: 2015-07-30 23:53:20,678 - INFO - Loaded plugin OBJWriter
30/07/15 23:53:20,681 Cura[15989]: 2015-07-30 23:53:20,680 - INFO - Loaded plugin STLWriter
30/07/15 23:53:20,684 Cura[15989]: 2015-07-30 23:53:20,683 - INFO - Loaded plugin GCodeWriter
30/07/15 23:53:20,687 Cura[15989]: 2015-07-30 23:53:20,686 - INFO - Loaded plugin CameraTool
30/07/15 23:53:20,690 Cura[15989]: 2015-07-30 23:53:20,690 - INFO - Loaded plugin MirrorTool
30/07/15 23:53:20,704 Cura[15989]: 2015-07-30 23:53:20,704 - INFO - Loaded plugin RotateTool
30/07/15 23:53:20,708 Cura[15989]: 2015-07-30 23:53:20,707 - INFO - Loaded plugin ScaleTool
30/07/15 23:53:20,710 Cura[15989]: 2015-07-30 23:53:20,710 - INFO - Loaded plugin SelectionTool
30/07/15 23:53:20,713 Cura[15989]: 2015-07-30 23:53:20,712 - INFO - Loaded plugin TranslateTool
30/07/15 23:53:20,716 Cura[15989]: 2015-07-30 23:53:20,716 - INFO - Checking for new version of cura
30/07/15 23:53:20,717 Cura[15989]: 2015-07-30 23:53:20,716 - INFO - Loaded plugin UpdateChecker
30/07/15 23:53:20,727 Cura[15989]: 2015-07-30 23:53:20,726 - INFO - Loaded plugin USBPrinting
30/07/15 23:53:20,733 Cura[15989]: 2015-07-30 23:53:20,732 - INFO - Loaded plugin CuraEngineBackend
30/07/15 23:53:21,309 Cura[15989]: 2015-07-30 23:53:21,309 - DEBUG - Loaded theme /Applications/Cura.app/Contents/Resources/cura/resources/themes/cura
30/07/15 23:53:21,739 Cura[15989]: file:///Applications/Cura.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
30/07/15 23:53:21,739 Cura[15989]: file:///Applications/Cura.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
30/07/15 23:53:22,101 Cura[15989]: 2015-07-30 23:53:22,100 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
30/07/15 23:53:22,102 Cura[15989]: 2015-07-30 23:53:22,102 - DEBUG - Backend connected on port 49675

Xisle, 2015-07-30:
Check out issue #154. Maybe there's something in there that can help you.

nallath, 2015-08-14:
Were you able to fix or is this still an issue?

EstebanPacheco, 2015-08-14:
Thanks for answering.
Well yes, but im still running into problems, for example if im printing smth and instart cura, the printer goes off and then on.

Best
Esteban

Thank

nallath, 2015-08-17:
Ok, but that is another issue.

closed 2015-08-17 by nallath