Ultimaker/Cura #204: Display corruption after resizing the window

fieldOfView, 2015-08-01:
After loading two objects, playing around with their scale and rotation, all while in Layer viewmode, I maximised the window. This led to the following display corruption:

[image]
It started more subtle (the 3d view showing through the settings panel), but in the screenshot I tried rotating the view for "extra effect".

Crash Handler did not catch the errors, but fortunately I was running Cura outside of Program Files so Cura.log could be written:
`

`

fieldOfView, 2015-08-01:
The errors in the above Cura.log may be caused by using the slider in Layers viewmode while the slicer is running and/or while the layers are loading.

awhiemstra, 2015-08-03:
Looks to me like your graphics driver gave up and stopped rendering properly. Unfortunately there is not much we can do about that. The other log errors are indeed related to event processing and should not lead to this kind of graphics corruption.

nickthetait, 2015-08-03:
The windows toolbar at the bottom is still rendering fine. So I would find it hard to blame the graphics driver.

fieldOfView, 2015-08-03:
I think @awhiemstra meant the opengl stack for the app, which is basically a part of the driver.

The computer this happened on is an otherwise well-behaving Core i5 with integrated HD Graphics 3000, and up-to-date drivers. I'll try to push my other computer, which has an nVidia card, to see if the same happens. If it does, we know it is not the driver.

nickthetait, 2015-08-03:
Aha! Thanks for doing more testing. :smile:

fieldOfView, 2015-08-03:
I have not yet reproduced this on other hardware. Then again, I can't reproduce it consistently on the Intel HD Graphics hw either.

What I do get on the nVidia hardware is that the shader for the UMO mesh doesn't work for a frame; I see a flash of black, but after that the rendering restores. So it seems that "something" fishy is going on during resize.

awhiemstra, 2015-08-04:
I was indeed referring to the OpenGL stack for the application.

I have seen this once with a colleague of mine, who also uses HD Graphics 3000 iirc. We also could never reproduce it since a restart of the application made things work normally again.

Unfortunately I have no NVidia hardware to test on so cannot easily reproduce the NVidia error. But maybe you can open a seperate issue for that?

closed 2015-09-15 by nallath