Ultimaker/Cura #206: Asymmetric viewport

fieldOfView, 2015-08-01:
The 3d viewport spans the full window-width, but the right part of the viewport is obscured by the settings panel. Effectively the settingspanel crops the 3d viewport, resulting in an asymmetric frustum. This looks weird.

Ideally, the width of the 3d view should be (window-width - sidebar-width).

awhiemstra, 2015-08-03:
Funnily enough, I have tried setting it that way several times but it always ends up feeling really weird to me. Admittedly, that is probably due to me being used to the current setup. However, the reason I never pushed any of those changes is that it breaks selection badly, so this is slightly trickier than just setting the viewport.

fieldOfView, 2015-08-03:
As an alternative to changing the viewport to be less wide, you could play around with the projection matrix to shift the view back to the center of the visible area (either directlty or using gl.glFrustum).

Admittedly, this is not an important issue at this point; there are more critical things to work on.

fieldOfView, 2015-09-15:
Looks like this got fixed: https://github.com/Ultimaker/Uranium/commit/094c5ae9e251d0d9f5f73f9733fe6bfc6fcbc25c
Thanks @awhiemstra!

closed 2015-09-15 by fieldOfView