Ultimaker/Cura #207: Zooming shortcut

alonsorobots, 2015-08-02:
Please add back the 'continuous' zooming navigation back.

With the new version of Cura we are limited do using 'incremental' zooming. Zooming in discrete steps is too slow and doesn't have the responsiveness that is useful for navigating 3D objects

My favorite 3D navigating scheme is:

Alt + right mouse click     = zooming
Alt + middle mouse click = panning
Alt + left mouse click       = pivoting around an object

If you feel like most of your target audience doesn't have a 3 button mouse, then use the scheme that I believe the old version of Cura used to use which is using two modifier keys like Ctrl + Shift or Alt + Shift. The best would be able to set this as a preference in cura for people that use a three button mouse since most people doing 3D modeling already have one.

Thanks,

-Alonso

nallath, 2015-08-10:
You can use + and - buttons for zooming already. We did some small experiments with the control buttons but they are not a very high priority at this moment.

alonsorobots, 2015-08-10:
Yup, I know I can use the + and - buttons already, but zooming, panning and orbiting are the bread and butter of navigation. I can tell you that when it's an interaction that you do every 15 seconds, it gets annoying quickly to have to press zoom out 10 times instead of holding down a button and click and dragging.

Thanks for all the great work on Cura. I can imagine the simple version will make it much more approachable for new users =)

nallath, 2015-08-11:
Fair enough. Also note that it is possible with the current architecture to create your own user interface plugins. It should even be possible to support space-mouses.

TMHogenhout, 2015-08-11:
It was an easy fix, so I added it for the 15.10 release. 

Double click is an absolute pain in the ass in Qt, so I added the functionality by using the space key and alt key as modifier keys. Beside that it pretty much works like Cura 15.4. 
[image]

TMHogenhout, 2015-08-11:
b107a1fe5e19bb7acdb0adbd97f40a541bd2b315

closed 2015-08-11 by TMHogenhout

alonsorobots, 2015-08-11:
You guys are the best!!  =)