Ultimaker/Cura #208: [cura_app.py] QWidget: Must construct a QApplication before a QWidget

thopiekar, 2015-08-02:
I get this error when trying to run the latest code from here (Github).
Hope you can give any suggestions while I'm using my own packaging, because running a standalone package also doesn't work for me.
If there is any interest into testing my packages, see https://launchpad.net/~thopiekar/+archive/ubuntu/wily-dev (along with other software)

However, this is the output I get:
```
thopiekar@home:~$ export LANG=C
thopiekar@home:~$ cura_app.py
QWidget: Must construct a QApplication before a QWidget
Aborted (core dumped)
```

awhiemstra, 2015-08-03:
You should be getting more console output than that. To me that suggests it cannot find any plugins. Looking at the package, you're not including any Uranium files and neither are you depending on Uranium. Since this is a hard dependency, you're going to need to solve that. See github.com/ultimaker/uranium for the code.

thopiekar, 2015-08-03:
I've got it installed manually, I just didn't cared much about the dependencies so far when just choosing the "cura" package.

Is there a way to make cura more verbose? So I could see where it expects the files 

Btw. I had to manually install the uranium scripts (I chose /usr/share/uranium/scripts for the) in order to build the cura package. I would love to see this to be done by cmake.

supergis, 2015-08-03:
I got same error.
This caused by UM not found. UM installed into /usr/local/python3/dist-packages,but normally may in  /usr/python3/dist-packages.
set PYTHONPATH= /usr/local/python3/dist-packages may let it run,but maybe got other error.

thopiekar, 2015-08-03:
You need to set your prefix to /usr when executing cmake. Just google for that.

I've got a clue what it could be. It might be related to Ubuntu's multiarch support, so the installed python interpreter is not looking into /usr/lib/python3/dist-packages/ it might suggest /usr/lib/$(DEB_MULTIARCH)/python3/dist-packages/ (eg. DEB_MULTIARCH = x86_64-linux-gnu for amd64).

thopiekar, 2015-08-03:
Ouch, I'm wrong. All arch-independent modules are installed into /usr/lib/python3/dist-packages/.
I wonder why I found Arcus here: /usr/lib/x86_64-linux-gnu/python3.4/dist-packages/Arcus/

supergis, 2015-08-03:
the gnu based lib will placed into /usr/lib/x86_64-linux-gnu/*.

thopiekar, 2015-08-03:
Just made a PR to fix the Python path for Arcus: https://github.com/Ultimaker/libArcus/pull/6

thopiekar, 2015-08-03:
Now more detailed information:

```
thopiekar@home:[...]$ python3
Python 3.4.3+ (default, Jul 28 2015, 13:17:50) 
[GCC 4.9.3] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cura.CuraApplication
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 4, in <module>
    from UM.Qt.QtApplication import QtApplication
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 17, in <module>
    from UM.Qt.Bindings.Bindings import Bindings
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/Bindings.py", line 14, in <module>
    from UM.Qt.Bindings.ControllerProxy import ControllerProxy
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/ControllerProxy.py", line 13, in <module>
    from UM.LoadWorkspaceJob import LoadWorkspaceJob
ImportError: No module named 'UM.LoadWorkspaceJob'
```

And it is true:
```
thopiekar@home:[...]$ ls /usr/lib/python3/dist-packages/UM/ -la
insgesamt 176
drwxr-xr-x 12 root root  4096 Aug  2 10:42 .
drwxr-xr-x 81 root root 12288 Aug  3 14:14 ..
-rw-r--r--  1 root root 13728 Aug  2 09:46 Application.py
drwxr-xr-x  3 root root  4096 Aug  2 10:42 Backend
-rw-r--r--  1 root root  1794 Aug  2 09:46 ColorGenerator.py
-rw-r--r--  1 root root  8806 Aug  2 09:46 Controller.py
-rw-r--r--  1 root root  3940 Aug  2 09:46 Event.py
-rw-r--r--  1 root root  1290 Aug  2 09:46 Extension.py
-rw-r--r--  1 root root  3428 Aug  2 09:46 i18n.py
-rw-r--r--  1 root root   183 Aug  2 09:46 __init__.py
-rw-r--r--  1 root root   454 Aug  2 09:46 InputDevice.py
-rw-r--r--  1 root root  3733 Aug  2 09:46 Job.py
-rw-r--r--  1 root root  3817 Aug  2 09:46 JobQueue.py
-rw-r--r--  1 root root  1799 Aug  2 09:46 Logger.py
drwxr-xr-x  3 root root  4096 Aug  2 10:42 Math
drwxr-xr-x  3 root root  4096 Aug  2 10:42 Mesh
-rw-r--r--  1 root root  2452 Aug  2 09:46 Message.py
drwxr-xr-x  3 root root  4096 Aug  2 10:42 Operations
drwxr-xr-x  3 root root  4096 Aug  2 10:42 OutputDevice
-rw-r--r--  1 root root   894 Aug  2 09:46 PluginError.py
-rw-r--r--  1 root root   542 Aug  2 09:46 PluginObject.py
-rw-r--r--  1 root root 11925 Aug  2 09:46 PluginRegistry.py
-rw-r--r--  1 root root  4177 Aug  2 09:46 Preferences.py
drwxr-xr-x  2 root root  4096 Aug  2 10:42 __pycache__
drwxr-xr-x  5 root root  4096 Aug  2 10:42 Qt
-rw-r--r--  1 root root  6558 Aug  2 09:46 Resources.py
drwxr-xr-x  4 root root  4096 Aug  2 10:42 Scene
-rw-r--r--  1 root root   766 Aug  2 09:46 SettingOverrideDecorator.py
drwxr-xr-x  4 root root  4096 Aug  2 10:42 Settings
-rw-r--r--  1 root root  6628 Aug  2 09:46 Signal.py
-rw-r--r--  1 root root  3579 Aug  2 09:46 Tool.py
drwxr-xr-x  3 root root  4096 Aug  2 10:42 View
```

awhiemstra, 2015-08-03:
Master had some breakage from the merge of the output_device branch. This should now be fixed.

thopiekar, 2015-08-03:
It is! Thank you very much!
Now it can't find any plugins. But will look whether it is already reported or needs to be opened.

closed 2015-08-03 by thopiekar