Ultimaker/Cura #209: Removing an object from the printbed does not trigger a reslice

fieldOfView, 2015-08-02:
* Load Ultimaker robot twice
* Scale both by 200%, so slicing takes a little while
Note that while changing the scale of the object, slicing will restart almost immediately
* While slicing of both objects is going on, remove one of the robots (by pressing backspace)
Note that slicing continues.
* After slicing is done, switch to Layers viewmode
Note that it shows layer for both robots, even if only 1 robot is currently loaded.

NB: while repro-ing these steps with 15.06.03, I also got the following "mix" of an unscaled robot with infill, and two scaled robots without infill:
[image]

fieldOfView, 2015-08-02:
Cura.log after doing this slice:
```
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ControllerProxy.py", line 26, in setActiveView
  File "c:\dev\uranium\UM\Controller.py", line 85, in setActiveView
  File "c:\dev\uranium\UM\Signal.py", line 94, in emit
  File "c:\dev\python34\lib\weakref.py", line 387, in items
RuntimeError: dictionary changed size during iteration
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ControllerProxy.py", line 26, in setActiveView
  File "c:\dev\uranium\UM\Controller.py", line 85, in setActiveView
  File "c:\dev\uranium\UM\Signal.py", line 94, in emit
  File "c:\dev\python34\lib\weakref.py", line 387, in items
RuntimeError: dictionary changed size during iteration
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 90, in mouseReleaseEvent
  File "c:\dev\uranium\UM\Qt\QtMouseDevice.py", line 36, in handleEvent
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Controller.py", line 211, in event
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\Tools\RotateTool\RotateTool.py", line 122, in event
    self.operationStopped.emit(self)
  File "c:\dev\uranium\UM\Signal.py", line 100, in emit
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\cura\cura\PlatformPhysics.py", line 115, in _onToolOperationStopped
  File "c:\dev\cura\cura\PlatformPhysics.py", line 108, in _onChangeTimerFinished
  File "c:\dev\uranium\UM\Operations\Operation.py", line 49, in push
  File "c:\dev\uranium\UM\Operations\OperationStack.py", line 34, in push
  File "c:\dev\cura\cura\PlatformPhysicsOperation.py", line 22, in redo
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 310, in setPosition
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 427, in _transformChanged
  File "c:\dev\uranium\UM\Signal.py", line 94, in emit
  File "c:\dev\python34\lib\weakref.py", line 387, in items
RuntimeError: dictionary changed size during iteration
Traceback (most recent call last):
  File "c:\dev\cura\cura\PlatformPhysics.py", line 108, in _onChangeTimerFinished
  File "c:\dev\uranium\UM\Operations\Operation.py", line 49, in push
  File "c:\dev\uranium\UM\Operations\OperationStack.py", line 34, in push
  File "c:\dev\cura\cura\PlatformPhysicsOperation.py", line 22, in redo
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 310, in setPosition
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 427, in _transformChanged
  File "c:\dev\uranium\UM\Signal.py", line 94, in emit
  File "c:\dev\python34\lib\weakref.py", line 387, in items
RuntimeError: dictionary changed size during iteration
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 90, in mouseReleaseEvent
  File "c:\dev\uranium\UM\Qt\QtMouseDevice.py", line 36, in handleEvent
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Controller.py", line 211, in event
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\Tools\RotateTool\RotateTool.py", line 122, in event
    self.operationStopped.emit(self)
  File "c:\dev\uranium\UM\Signal.py", line 100, in emit
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\cura\cura\PlatformPhysics.py", line 115, in _onToolOperationStopped
  File "c:\dev\cura\cura\PlatformPhysics.py", line 108, in _onChangeTimerFinished
  File "c:\dev\uranium\UM\Operations\Operation.py", line 49, in push
  File "c:\dev\uranium\UM\Operations\OperationStack.py", line 34, in push
  File "c:\dev\cura\cura\PlatformPhysicsOperation.py", line 22, in redo
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 310, in setPosition
  File "c:\dev\uranium\UM\Scene\SceneNode.py", line 427, in _transformChanged
  File "c:\dev\uranium\UM\Signal.py", line 94, in emit
  File "c:\dev\python34\lib\weakref.py", line 387, in items
RuntimeError: dictionary changed size during iteration
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 90, in mouseReleaseEvent
  File "c:\dev\uranium\UM\Qt\QtMouseDevice.py", line 36, in handleEvent
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Controller.py", line 211, in event
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\Tools\RotateTool\RotateTool.py", line 122, in event
    self.operationStopped.emit(self)
  File "c:\dev\uranium\UM\Signal.py", line 100, in emit
[... 232 lines cut ...]
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 386
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 383
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 380
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\LayerView\LayerView.py", line 98, in beginRendering
    layer_mesh = layer_data.getLayer(layer).createMesh()
  File "C:\Users\Aldo\Desktop\Cura\Cura_15.06.03\plugins\CuraEngineBackend\LayerData.py", line 31, in getLayer
    return self._layers[layer]
KeyError: 377
```

awhiemstra, 2015-08-03:
Sounds like we're missing a proper handling of childrenChanged somewhere. Thank you for reporting it, I will look into it.

nallath, 2015-08-21:
It seems that the delete does not properly trigger the childrenChanged

closed 2015-08-24 by nallath