Ultimaker/Cura #212: dynamic Machine Settings?

gearsawe, 2015-08-02:
I know this is not right but this is what I am trying to do. Need to be able to have some machine settings be dynamic for example machine_disallowed_areas. For the sake of testing I chose layer_height to see it this would even work. Get string to float error or I remove quotes get expected ","  after the "[]"
```
{
    "id": "M3D_Micro_Adj",
    "name": "M3D Micro Adj",
    "icon": "icon_ultimaker.png",
    "platform": "M3D_Micro_platform.stl",
    "inherits": "M3D_Micro_Short.json",
    "machine_settings": {
        "machine_disallowed_areas": { 
            "default": "[] if Layer Height > 0.1 else [[[0,0],[0,20],[20,20],[20,0]]]"
        }
    },
    "categories": {
        "material": {
            "label": "Material",
            "visible": true,
            "icon": "category_material",
            "settings": {
                "layer_height": {
                    "label": "Layer Height",
                    "description": "blah blah",
                    "unit": "mm",
                    "type": "float",
                    "default": 0.1,
                    "min_value": 0.0001,
                    "min_value_warning": 0.04,
                    "max_value_warning": 0.32
                }
            }
        }
    }
}
```

awhiemstra, 2015-08-03:
The default value should always be a constant. In this case, it is a string and since other areas expect the value to be an array you get errors when it is not. The only setting that is allowed to use expressions at the moment is inherit_function. You can try using that to calculate the value, but I am not sure if they are being evaluated for machine settings.

gearsawe, 2015-08-03:
@awhiemstra . Thanks for taking a look at this, I was kind afraid this was not possible. This would have been great if it worked. Now I need two printer configurations for the same printer which is not very convenient at all. I tried the inherit_function. I don't get any errors but nothing happens.

Would this be worth a feature request? Others might find his useful. Basically in my case the foot print changes because the frame of the printer gets in the way of the head. But there could be other settings many would find more useful if they were dynamic.

awhiemstra, 2015-08-04:
While a feature request for it is fine, you might also be interested in the variant handling we are introducing for 15.10, which might also be useful for your situation. It should allow quick selection of a machine with slightly different machine settings once we are fully done implementing it. The current idea is that you can have two separate JSON files with the same ID that will be treated as variants, so you can change any setting defaults in those files.

closed 2015-08-10 by nallath