Ultimaker/Cura #215: RuntimeError: Could not load the backend plugin!

thopiekar, 2015-08-03:
```
Version: master
Platform: Linux-4.1.0-2-generic-x86_64-with-Ubuntu-15.10-wily
Qt: 5.4.2
PyQt: 5.4.2

Exception:
Traceback (most recent call last):
  File "/usr/bin/cura_app.py", line 9, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 301, in getInstance
    Application._instance = cls()
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 58, in __init__
    super().__init__(name = "cura", version = "master")
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 73, in __init__
    self._loadPlugins()
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 109, in _loadPlugins
    raise RuntimeError("Could not load the backend plugin!")
RuntimeError: Could not load the backend plugin!
```

Don't know which directory Cura expects but here are the installed files:
```
/usr/lib/cura/plugins
/usr/lib/cura/plugins/CuraEngineBackend*
/usr/lib/cura/plugins/GCodeWriter*
/usr/lib/cura/plugins/LayerView*
/usr/lib/cura/plugins/RemovableDriveOutputDevice*
/usr/lib/cura/plugins/USBPrinting*
/usr/lib/uranium
/usr/lib/uranium/plugins
/usr/lib/uranium/plugins/ConsoleLogger*
/usr/lib/uranium/plugins/FileHandlers*
/usr/lib/uranium/plugins/FileLogger*
/usr/lib/uranium/plugins/LocalFileOutputDevice*
/usr/lib/uranium/plugins/Tools
/usr/lib/uranium/plugins/Tools/CameraTool*
/usr/lib/uranium/plugins/Tools/MirrorTool*
/usr/lib/uranium/plugins/Tools/RotateTool*
/usr/lib/uranium/plugins/Tools/ScaleTool*
/usr/lib/uranium/plugins/Tools/SelectionTool*
/usr/lib/uranium/plugins/Tools/TranslateTool*
/usr/lib/uranium/plugins/UpdateChecker*
/usr/lib/uranium/plugins/Views
/usr/lib/uranium/plugins/Views/MeshView*
/usr/lib/uranium/plugins/Views/WireframeView*
```

awhiemstra, 2015-08-04:
Please check the Cura log. It probably produces an error about not finding either Arcus or google.protobuf when trying to load the backend. Since Cura makes little sense without the backend we produce an exception when that happens.

thopiekar, 2015-08-04:
Yes, you were guessing it right. It can't find:
```
ERROR - Import error loading module CuraEngineBackend: No module named 'Arcus'
```
I think the PR I sent yesterday will fix this problem, as my Arcus installation goes to /usr/lib/x86_64-linux-gnu/python3.4/dist-packages/Arcus/ except of /usr/lib/python3.4/dist-packages/Arcus/

See: https://github.com/Ultimaker/libArcus/pull/6

thopiekar, 2015-08-07:
Fixed by /libArcus#6

closed 2015-08-07 by thopiekar