Ultimaker/Cura #217: Cura 15.06.03 cuts my part in half.

KepecsLab, 2015-08-04:
My part is fine in the solid view:

[image]

but Cura cuts it in half in the layers view:

[image]

When supports are permitted, it fills the gap with support material.

I tried to print and verified that this is not just a display error - it prints the part with a gap.
I loaded the part in Cura 15.06.02, same problem.
I'm running Cura on Windows 7 SP1, with all updates, using Bootcamp on a Macbook Pro (2013).

BagelOrb, 2015-08-05:
Your model is probably incorrect. It may have faces on the inside or too large gaps between faces which are supposed to be touching. 

In the old cura this could be viewed in the x ray view, but this hasn't been implemented yet in the new cura.

closed 2015-08-05 by BagelOrb

KepecsLab, 2015-08-05:
Yes, this worked - I found a few stray lines in my CAD software, removed them, and now it slices properly.

It would be nice if future versions of Cura showed the missing space in the solid view as well, so it's obvious that something is wrong immediately on load.