Ultimaker/Cura #218: Print parameters not fully transmited to printer

pascalmayer, 2015-08-05:
Print temperature, builder plate temeprature and fan speed are not set by the gcode file (Ultmaker 2).
Thanks

BagelOrb, 2015-08-05:
Indeed, they are set by the printer itself.

closed 2015-08-05 by BagelOrb

pascalmayer, 2015-08-05:
Sure, but this prevents from having “print ready” gcode files with optimal settings for different objects.

It would help a lot, especially when the default printer settings are not optimal: I have to readjust them at each print, thus opening the door for mistakes.

Also, if the printing is done by a non-specialist, it would help a lot and reduce the risk of errors.

Pascal Mayer

BagelOrb, 2015-08-06:
Thing is that the temperature depends foremost on the material in the printer, while cura doesn't know what material is in the printer; only the printer knows.

We would like to find some filament independent temperature control statistics, so that they can be combined with the filament temperature statistics of the printer, but we haven't accomplished this yet.

Now I think of it, you tweak the temperature to some goal. I suppose what you want to tweak is the viscosity of the filament. Is this the only goal you have for tweaking temperature? 
Given this goal, you could be able to set the viscosity in cura while the printer translates that to temperature settings via a viscosity plot it has on each material.

pascalmayer, 2015-08-06:
Dear Tim,

 

thanks for the feedback. I’d love to set the viscosity, but I couldn’t find the setting in the advanced options. (BTW, the scroll bar of the option window is extremely thin, not easy to use or “bug” on Windows 8.1, see attached capture).

 

As I’m myself busy with instrument development, I dare to suggest that the printer checks if its internal setting on material is matching the indication of the gcode file, and if different, imposes temperature to be set on the printer (presently, the temperature sets itself to 0 anyhow each time I start a print), otherwise, uses the settings of the gcode file ? Could be left as an advanced option.

 

Actually, I start wondering if the temperature control of the printer is doing ok as I’m unable to print anything in PLA at 210°C and have to rise temperature at 225°C to start getting acceptable results, and could not print in ABS so far (but only struggling since 10 days, so not sure yet if I’m the one to be blamed or the printer). Same for the bed temperature, no way anything would stick at 60°C, need to go to 75°C, and still far from perfect (I re-adjusted bed horizontality, cleaned etc… Will retry the whole process today again) .

 

Thanks again and regards,

 

Pascal.

 

 

 

Dr Pascal Mayer, PhD

CEO

Alphanosos S.A.S.

20-22 rue Henri et Gilberte Goudier, 63200 Riom, France

 <http://www.alphanosos.com> www.alphanosos.com

+33 659 856 881 / +33 982 216 506

SkypeID pascal.mayer.fr63

BagelOrb, 2015-08-06:
The attachment didn't get sent along, because these messages are going via github. Please make a separate issue for the sidebar and post the image there.

The viscosity idea is not yet implemented and - given the extent of testing we need to perform to get all the plots - won't be any time soon.

It sounds like your nozzle is clogged. Try performing the atomic method.