Ultimaker/Cura #219: Crash on startup

BagelOrb, 2015-08-05:
Version: master
Platform: Linux-3.19.0-25-generic-x86_64-with-Ubuntu-15.04-vivid
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 16, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "/home/tim/Development/Uranium/UM/Application.py", line 329, in getInstance
    Application._instance = cls()
  File "/home/tim/Development/Cura/cura/CuraApplication.py", line 58, in __init__
    super().__init__(name = "cura", version = "master")
  File "/home/tim/Development/Uranium/UM/Qt/QtApplication.py", line 73, in __init__
    self._loadPlugins()
  File "/home/tim/Development/Cura/cura/CuraApplication.py", line 106, in _loadPlugins
    self._plugin_registry.loadPlugins()
TypeError: loadPlugins() missing 1 required positional argument: 'meta_data'

Didn't have the issue at commit 7c2617133e30a20b2b073db901a6a2431c945dea

BagelOrb, 2015-08-05:
Last commit known to have the issue: 9c625a7c41d48cd8641644154937240fd676920d

awhiemstra, 2015-08-05:
Uranium API has changed to accommodate for this. Make sure you update Uranium to recent master as well.

closed 2015-08-05 by awhiemstra