Ultimaker/Cura #220: Cura does not create ultiflavor gcode

Bas-van-Deursen, 2015-08-05:
Application Version: 15.06.03
Platform:Windows 7 64-bit

Display Driver: NVIDEA Quattro

Steps to Reproduce:

Selected printer: UM2

slicing a model with advanced settings and save to sd card. The generated gcode does not have the right starting lines for UM2 gcode:

;Generated with Cura_SteamEngine DEV
;LAYER:0
M107
G0 F9000 X56.243 Y76.312 Z0.299
G0 X58.764 Y75.628
;TYPE:SKIRT

Bas-van-Deursen, 2015-08-05:
Tried to replicate the problem, but I can't get it to fail again... Now it does work.

awhiemstra, 2015-08-05:
Seems like prepending the start GCode failed. I wonder how that can happen. If it happens again, can you save and post the engine log?

Bas-van-Deursen, 2015-08-05:
will do.

-Bas

-- 

*Bas van Deursen *Design Engineer
Ultimaker B.V.

*Watermolenweg 2, 4191 PN, Geldermalsen*
*T *+31 (0)345 712017 | *E *b.vandeursen@ultimaker.com | *W *
www.ultimaker.com

closed 2015-08-10 by nallath