Ultimaker/Cura #221: UM.PluginError.PluginNotFoundError: Could not find plugin ConsoleLogger

supergis, 2015-08-05:
Now,Ubuntu15.04 got another error:
```
Version: master
Platform: Linux-3.19.0-25-generic-x86_64-with-Ubuntu-15.04-vivid
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 16, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 301, in getInstance
    Application._instance = cls()
  File "/home/supermap/dev/Cura/cura/CuraApplication.py", line 58, in __init__
    super().__init__(name = "cura", version = "master")
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 73, in __init__
    self._loadPlugins()
  File "/home/supermap/dev/Cura/cura/CuraApplication.py", line 104, in _loadPlugins
    self._plugin_registry.loadPlugin("ConsoleLogger")
  File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 69, in loadPlugin
    raise PluginNotFoundError(id)
UM.PluginError.PluginNotFoundError: Could not find plugin ConsoleLogger
```

awhiemstra, 2015-08-05:
ConsoleLogger is provided by Uranium. This suggests it cannot load any Uranium plugins.

supergis, 2015-08-06:
I trace the plugin folder,got:
```
Loading plugin:ConsoleLogger
Find Plugin:/home/supermap/dev/lib/uranium
Find Plugin:/usr/bin/plugins
Find Plugin:/home/supermap/dev/Resources/uranium/plugins
Find Plugin:/home/supermap/dev/Resources/cura/plugins
Find Plugin:/home/supermap/.local/share/cura/plugins
Find Plugin:/usr/lib/python3/dist-packages/UM/../plugins
Find Plugin:/home/supermap/dev/lib/cura
Find Plugin:/home/supermap/dev/Cura/cura/../plugins
```
But "ConsoleLogger" in the Uranium/Plugins actually. I copy it into Cura/plugins,now work.
I think this isn't good practice,May place the directory Under UM,like /usr/lib/python3/dist-packages/UM/../plugins,and merge all plugins into this directory.
Would anyone give some suggestions?

nallath, 2015-08-10:
It should use more plug-ins from Uranium than just the console logger. It's a bit weird that it only has problems with the console one.

awhiemstra, 2015-08-10:
It errors out on ConsoleLogger because I added code to master that explicitly loads it before other plugins.

As to where it looks for plugins, many of them are relative to the application install prefix. The install prefix is determined by assuming either the python binary or the application binary is located in a subdirectory of the prefix. If this process fails I can imagine it not finding plugins properly.

closed 2015-08-14 by nallath

mariolpantunes, 2018-03-01:
I have the same exact error with Cura 3.2.1 and running on Linux (Slackware 14.2).
I am able to fix it by moving the cura and uranium plugins to /usr/lib instead of /usr/lib64.

Ghostkeeper, 2018-03-23:
We don't store the plug-ins in `/usr/lib` though. Where did you get your Cura distribution from?

We distribute Cura via an [.AppImage file](https://github.com/Ultimaker/Cura/releases/tag/3.2.1) to Linux users. Inside the files the plug-ins are stored in `/usr/bin/plugins/plugins`.

mariolpantunes, 2018-03-26:
I compile the code from source.

Ghostkeeper, 2018-04-19:
Then it'll look for the plug-ins in `<Application.py>/../plugins` for Uranium and in `<CuraApplication.py>/../plugins` for Cura.