Ultimaker/Cura #222: Build CuraEngine failed.

supergis, 2015-08-05:
On Ubuntu15.04, Build CuraEngine got error:
```
Linking CXX executable CuraEngine
CMakeFiles/CuraEngine.dir/src/commandSocket.cpp.o：在函数‘google::protobuf::internal::GenericTypeHandler<cura::proto::SlicedObject>::New(google::protobuf::Arena*)’中：
commandSocket.cpp:(.text._ZN6google8protobuf8internal18GenericTypeHandlerIN4cura5proto12SlicedObjectEE3NewEPNS0_5ArenaE[_ZN6google8protobuf8internal18GenericTypeHandlerIN4cura5proto12SlicedObjectEE3NewEPNS0_5ArenaE]+0x73)：对‘google::protobuf::Arena::AllocateAligned(unsigned long)’未定义的引用
```
Would any suggestion for me? Thanks in advance.

thopiekar, 2015-08-06:
Which version of libprotobuf are you using? You'll needed a very recent version. I use in my PPA a version >=3.0.0.

thopiekar, 2015-08-06:
I just created a seperated PPA now for Cura works now here for me on Ubuntu Wily (development version). It should be complete shortly: https://launchpad.net/~thopiekar/+archive/ubuntu/cura

There are still things to fix, like removing updater_plugin and fixing icon image for the application menu entry, but it is usable :smiley:

nallath, 2015-08-10:
This is an Cura Engine issue. Please post it there.

closed 2015-08-10 by nallath