Ultimaker/Cura #228: Concentric Infill Enhancement

Xisle, 2015-08-06:
I am really liking the concentric infill so far! That's all I've been using.

I do have an issue with how the infill is generated from outside to center sometimes and center to outside others. When the infill starts in the center of a section it has nothing to connect to. It would be best if the concentric "rings" were always built from outside to center.

BagelOrb, 2015-08-07:
This should be fairly easy to fix. Will look into it..

nallath, 2015-08-14:
This is actually an CuraEngine issue, so closing it here.

BagelOrb, 2015-08-17:
So did you open the issue there then?

nallath, 2015-08-17:
No, but it's not a Cura issue, so it's up to the orignal poster to change it as otherwise they don't get the status updates if anyone posts there.

BagelOrb, 2015-11-04:
This is a duplicate of https://github.com/Ultimaker/CuraEngine/issues/235

closed 2015-11-04 by BagelOrb