Ultimaker/Cura #229: [master] WARNING - Could not find any translations matching default for file uranium_qt, falling back to english

thopiekar, 2015-08-07:
Would be interesting where Cura expects it's i18n files. However, this is the message I get here:
```
2015-08-07 13:11:30,584 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-08-07 13:11:30,585 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-08-07 13:11:30,587 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-08-07 13:11:30,587 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
```
And the location of the files, while installing the files via CMake:
```
thopiekar@home:~$ ls /usr/share/cura/resources/i18n/ -la                                                                                                               
insgesamt 84                                                                                                                                                           
drwxr-xr-x 7 root root  4096 Aug  6 22:32 .                                                                                                                            
drwxr-xr-x 8 root root  4096 Jul 31 23:09 ..                                                                                                                           
-rw-r--r-- 1 root root  2818 Aug  6 18:51 cura.pot                                                                                                                     
-rw-r--r-- 1 root root  8515 Aug  6 18:51 cura_qt.pot                                                                                                                  
drwxr-xr-x 2 root root  4096 Aug  6 22:32 de
drwxr-xr-x 2 root root  4096 Aug  6 22:32 es
-rw-r--r-- 1 root root 40688 Aug  6 18:51 fdmprinter.json.pot
drwxr-xr-x 2 root root  4096 Aug  6 22:32 fi
drwxr-xr-x 2 root root  4096 Aug  6 22:32 fr
drwxr-xr-x 2 root root  4096 Aug  6 22:32 ru
```
Same for uranium:
```
thopiekar@home:~$ ls /usr/share/uranium/resources/i18n -la
insgesamt 48
drwxr-xr-x 7 root root 4096 Aug  5 18:36 .
drwxr-xr-x 5 root root 4096 Aug  3 18:17 ..
drwxr-xr-x 2 root root 4096 Aug  5 18:36 de
drwxr-xr-x 2 root root 4096 Aug  5 18:36 es
drwxr-xr-x 2 root root 4096 Aug  5 18:36 fi
drwxr-xr-x 2 root root 4096 Aug  5 18:36 fr
drwxr-xr-x 2 root root 4096 Aug  5 18:36 ru
-rw-r--r-- 1 root root 8870 Aug  5 18:16 uranium.pot
-rw-r--r-- 1 root root 5691 Aug  5 18:16 uranium_qt.pot
```

awhiemstra, 2015-08-07:
It is probably missing the compiled binary files. The CMake build files contain targets for these files, once compiled they are excepted to be located in $resources/i18n/$lang/LC_MESSAGES

thopiekar, 2015-08-07:
Doesn't look like it is done by CMake:

```
thopiekar@home:~$ ls -la /usr/share/cura/resources/i18n/de/
insgesamt 92
drwxr-xr-x 2 root root  4096 Aug  6 22:32 .
drwxr-xr-x 7 root root  4096 Aug  6 22:32 ..
-rw-r--r-- 1 root root  3264 Aug  6 18:51 cura.po
-rw-r--r-- 1 root root  9934 Aug  6 18:51 cura_qt.po
-rw-r--r-- 1 root root 67988 Aug  6 18:51 fdmprinter.json.po
```
```
thopiekar@home:~$ ls -la /usr/share/uranium/resources/i18n/de/
insgesamt 28
drwxr-xr-x 2 root root  4096 Aug  5 18:36 .
drwxr-xr-x 7 root root  4096 Aug  5 18:36 ..
-rw-r--r-- 1 root root 10202 Aug  5 18:16 uranium.po
-rw-r--r-- 1 root root  6387 Aug  5 18:16 uranium_qt.po
```
Additonally the build logs:
 * cura: https://launchpadlibrarian.net/213778154/buildlog_ubuntu-wily-amd64.cura_0.0.0-0~201508061651~rev275~pkg12~ubuntu15.10.1_BUILDING.txt.gz
 * uranium: https://launchpadlibrarian.net/213776928/buildlog_ubuntu-wily-amd64.uranium_0.0.0-0~201508061633~rev865~pkg12~ubuntu15.10.1_BUILDING.txt.gz

awhiemstra, 2015-08-07:
It is not done by the default target, but by the translations target. Additionally, it currently lacks install commands since I did not want to have to depend on gettext and QtLinguistTools at build time. I can add them, or you can use https://github.com/Ultimaker/cura-binary-data which contains pre-compiled binaries with the needed data.

thopiekar, 2015-08-07:
```
Additionally, it currently lacks install commands since I did not want to have to depend on gettext and QtLinguistTools at build time.
```
No problem. So when gettext and QtLinguistTools are available/installed the missing files get generated, right?

awhiemstra, 2015-08-07:
It does now. :) Unfortunately it requires a bit more work as the file install only works after a second run of CMake.

thopiekar, 2015-12-22:
Now I'm on 2.1 and i18n gets installed on first CMake run. 
But the language files are still not used as the UI is still in english. Moreover I don't see a error message in the output of cura_app.py.

Here they are:
```
/usr/share/cura
/usr/share/cura/resources
/usr/share/cura/resources/i18n
/usr/share/cura/resources/i18n/cura.pot
/usr/share/cura/resources/i18n/de
/usr/share/cura/resources/i18n/de/cura.po
/usr/share/cura/resources/i18n/de/fdmprinter.json.po
/usr/share/cura/resources/i18n/fdmprinter.json.pot
/usr/share/cura/resources/i18n/fi
/usr/share/cura/resources/i18n/fi/cura.po
/usr/share/cura/resources/i18n/fi/fdmprinter.json.po
/usr/share/cura/resources/i18n/fr
/usr/share/cura/resources/i18n/fr/cura.po
/usr/share/cura/resources/i18n/fr/fdmprinter.json.po
/usr/share/cura/resources/i18n/pl
/usr/share/cura/resources/i18n/pl/cura.po
/usr/share/cura/resources/i18n/pl/fdmprinter.json.po
```

thopiekar, 2015-12-22:
Now tested it with running cmake twice, but same result. I see in the logs that the mo files get built, but are just not copied when doing ```make install```. So running cmake twice makes no difference.

Build, which runs cmake twice: https://launchpadlibrarian.net/231106496/buildlog_ubuntu-wily-amd64.cura_1%3A2.1.0-0~201512222210~rev515~pkg27~ubuntu15.10.1_BUILDING.txt.gz
Build, which runs cmake once: https://launchpadlibrarian.net/231001143/buildlog_ubuntu-vivid-amd64.cura_1%3A2.1.0-0~201512221116~rev515~pkg26~ubuntu15.04.1_BUILDING.txt.gz

thopiekar, 2015-12-23:
Found a solution for this and will add a PR soon.
Hope you'll like it :smile:

thopiekar, 2016-03-10:
Merged and fixed

closed 2016-03-10 by thopiekar