Ultimaker/Cura #230: Cannot rotate, mirror or scale in 15.06

BESPOKE3DPRINTING, 2015-08-07:
The latest version isn't allowing me to rotate, scale or mirror any of my .stl files
any ideas?
Kind regards

nallath, 2015-08-10:
Please have a look at https://github.com/Ultimaker/Cura#logging-issues. We need more information before we can help.

BagelOrb, 2015-08-10:
Besides the information @nallath is asking for:
do you get the rotate, scale etc. buttons on the bottom left?
Do you get the on-model arrows in red, green and blue?

BESPOKE3DPRINTING, 2015-08-10:
I can see the buttons on the left but I am not seeing the on model co
ordinate system. It's like cura isn't recognising any of the stl files.

awhiemstra, 2015-08-10:
Can you select objects by clicking on them?

BESPOKE3DPRINTING, 2015-08-10:
No

awhiemstra, 2015-08-10:
Then it is a duplicate of #84

closed 2015-08-10 by awhiemstra

BESPOKE3DPRINTING, 2015-08-10:
Ok I'll try that.
Thanks for your support.