Ultimaker/Cura #231: Default object size not correct when inserted with build platform and disallow areas. BUG

gearsawe, 2015-08-09:
This is really a odd bug
two printer settings Tiko and Tiko2. 
Tiko- object automatticaly scales up to max size and making it too big
[image]

```
{
    "id": "Tiko",
    "name": "Tiko",
    "icon": "icon_ultimaker.png",
    "inherits": "fdmprinter.json",
    "machine_settings": {
        "machine_center_is_zero": { "default": true },
        "machine_heated_bed": { "default": false },
        "machine_gcode_flavor": { "default": "RepRap" },
        "machine_width": { "default": 185.1 },
        "machine_depth": { "default": 185.1 },
        "machine_height": { "default": 125 },
        "machine_nozzle_size": { "default": 0.4 },
        "machine_head_shape_min_x": { "default": 75 },
        "machine_head_shape_min_y": { "default": 18 },
        "machine_head_shape_max_x": { "default": 18 },
        "machine_head_shape_max_y": { "default": 35 },
        "machine_disallowed_areas": { "default": [
        [[-92.55,-92.55],[0.00,-92.55],[-16.22,-88.21],[-92.55,-88.21]],
        [[-92.55,-88.21],[-16.22,-88.21],[-28.10,-76.33],[-92.55,-76.33]],
        [[-92.55,-76.33],[-28.10,-76.33],[-80.15,13.83],[-92.55,13.83]],
        [[-92.55,13.83],[-80.15,13.83],[-84.50,30.05],[-92.55,30.05]],
        [[-92.55,30.05],[-84.50,30.05],[-80.15,46.28],[-92.55,46.28]],
        [[-92.55,46.28],[-80.15,46.28],[-68.28,58.15],[-92.55,58.15]],
        [[-92.55,58.15],[-68.28,58.15],[-52.05,62.50],[-92.55,62.50]],
        [[-92.55,62.50],[92.55,62.50],[92.55,92.55],[-92.55,92.55]],
        [[92.55,-92.55],[0.00,-92.55],[16.22,-88.21],[92.55,-88.21]],
        [[92.55,-88.21],[16.22,-88.21],[28.10,-76.33],[92.55,-76.33]],
        [[92.55,-76.33],[28.10,-76.33],[80.15,13.83],[92.55,13.83]],
        [[92.55,13.83],[80.15,13.83],[84.50,30.05],[92.55,30.05]],
        [[92.55,30.05],[84.50,30.05],[80.15,46.28],[92.55,46.28]],
        [[92.55,46.28],[80.15,46.28],[68.28,58.15],[92.55,58.15]],
        [[92.55,58.15],[68.28,58.15],[52.05,62.50],[92.55,62.50]]
        ]}
    },
    "categories": {
        "material": {
            "settings": {
                "material_bed_temperature": {
                    "visible": false
                },
                "layer_height": {
                    "label": "Layer Height",
                    "description": "blah",
                    "unit": "mm",
                    "type": "float",
                    "default": 0.1,
                    "min_value": 0.0001,
                    "min_value_warning": 0.04,
                    "max_value_warning": 0.32
                }
            }
        }
    }
}
```
Tiko2- no disallow areas and the object is correct size when inserted. very odd?
[image]

```
{
    "id": "Tiko2",
    "name": "Tiko2",
    "icon": "icon_ultimaker.png",
    "inherits": "fdmprinter.json",
    "machine_settings": {
        "machine_center_is_zero": { "default": true },
        "machine_heated_bed": { "default": false },
        "machine_gcode_flavor": { "default": "RepRap" },
        "machine_width": { "default": 185.1 },
        "machine_depth": { "default": 185.1 },
        "machine_height": { "default": 125 },
        "machine_nozzle_size": { "default": 0.4 },
        "machine_head_shape_min_x": { "default": 75 },
        "machine_head_shape_min_y": { "default": 18 },
        "machine_head_shape_max_x": { "default": 18 },
        "machine_head_shape_max_y": { "default": 35 },
        "machine_disallowed_areas": { "default": [
        ]}
    },
    "categories": {
        "material": {
            "settings": {
                "material_bed_temperature": {
                    "visible": false
                },
                "layer_height": {
                    "label": "Layer Height",
                    "description": "Blah",
                    "unit": "mm",
                    "type": "float",
                    "default": 0.1,
                    "min_value": 0.0001,
                    "min_value_warning": 0.04,
                    "max_value_warning": 0.32
                }
            }
        }
    }
}

nallath, 2015-08-10:
I believe that the scale to max does not use the disallowed areas to scale but the bounding box. We need to fix that.

gearsawe, 2015-08-10:
@nallath I know the bug you talking about. But what I don't understand is why it is automatically scaling the object in the first image. it should be like the second image smaller. 

I have another case when an object is inserted it turns up to microscopic but fine when there are no disallow areas. The work around right now for me is switch printer configs insert the object then switch back. 

I went another step. removed all the but a couple of the disallow areas and the objects inserts fine at the correct scale. so currently the disallow areas are affecting the object size but almost in a completely random manner. It is something to do with shape, area bed size or something, not sure yet. All I do know is the shape of the disallow area and size of the bed I have above image it messing something up and why?

on the last test i reduce the size of the print bed to 100x100 keeping the disallow areas and then object inserts fine. So now I am really confused. The auto-scaling when inserting is messing things up.

nallath, 2015-08-11:
Right, that is weird. I wil discuss this with @awhiemstra, as we need to think about a better way for non-square plates anyway.

BagelOrb, 2017-07-21:
This issue is so old that I cannot imagine this still happens.

closed 2017-07-21 by BagelOrb

Ghostkeeper, 2017-07-27:
We don't autoscale any more except in factors of 1000, and scale to max is removed too.