Ultimaker/Cura #232: Question: How do I get the option for USB printing to appear?

KJHamblin, 2015-08-09:
I have a RigidBotBig based on the Arduino Mega 2560 and now that I have my .json profiles complete I cannot get Cura 15.06.03 to give me the option to "Print with USB".  The only option that appears is "Save to Disk."  This is despite the fact that the printer shows up in Device Manager and connects to Pronterface and Cura 15.04 fine.  This is true on both Windows 7 and Windows 8.1.

Is it supported yet or am I doing something incorrect?

nallath, 2015-08-10:
USB printing tries to send a number of commands to the printer. Based on the response that it gets it decides if the printer is in fact something it can talk to. I've only tested this on UM machines as those are the only ones i have acces to. 
Could you post the log file? This should have some logs about trying (and failing) to acces the printer.

KJHamblin, 2015-08-10:
Here are the contents of the log file.  Please ignore the issue with the backplate image.  I was toying around with that on another machine.  I haven't gotten the backplate to work, but that is unrelated and much less important than the "Print with USB" option.

There was a serial time out: serial.serialutil.SerialTimeoutException: Write timeout

```
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\AvailableMachinesModel.py", line 38, in createMachine
  File "c:\dev\uranium\UM\Application.py", line 246, in setActiveMachine
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Scene\Platform.py", line 52, in _onActiveMachineChanged
  File "c:\dev\uranium\UM\Resources.py", line 50, in getPath
FileNotFoundError: Could not find resource ('RigidBotBigbackplate.png',) in 7
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MachinesModel.py", line 32, in removeMachine
IndexError: list index out of range
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\AvailableMachinesModel.py", line 38, in createMachine
  File "c:\dev\uranium\UM\Application.py", line 246, in setActiveMachine
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Scene\Platform.py", line 52, in _onActiveMachineChanged
  File "c:\dev\uranium\UM\Resources.py", line 50, in getPath
FileNotFoundError: Could not find resource ('RigidBotBigbackplate.png',) in 7
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MachinesModel.py", line 32, in removeMachine
  File "c:\dev\uranium\UM\Application.py", line 246, in setActiveMachine
  File "c:\dev\uranium\UM\Signal.py", line 96, in emit
  File "c:\dev\uranium\UM\Scene\Platform.py", line 52, in _onActiveMachineChanged
  File "c:\dev\uranium\UM\Resources.py", line 50, in getPath
FileNotFoundError: Could not find resource ('RigidBotBigbackplate.png',) in 7
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\Views\MeshView\MeshView.py", line 62, in beginRendering
    if not node.render(renderer):
  File "c:\dev\uranium\UM\Scene\Platform.py", line 32, in render
  File "c:\dev\uranium\UM\Resources.py", line 50, in getPath
FileNotFoundError: Could not find resource ('RigidBotBigbackplate.png',) in 7
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\MainWindow.py", line 138, in _render
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\Views\MeshView\MeshView.py", line 62, in beginRendering
    if not node.render(renderer):
  File "c:\dev\uranium\UM\Scene\Platform.py", line 32, in render
  File "c:\dev\uranium\UM\Resources.py", line 50, in getPath
FileNotFoundError: Could not find resource ('RigidBotBigbackplate.png',) in 7
Exception in thread Thread-34:
Traceback (most recent call last):
  File "c:\dev\python34\lib\threading.py", line 921, in _bootstrap_inner
  File "c:\dev\python34\lib\threading.py", line 869, in run
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\USBPrinting\PrinterConnection.py", line 213, in _connect
    self._serial.write(b"\n")
  File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 295, in write
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\USBPrinting\PrinterConnection.py", line 312, in _sendCommand
    self._serial.write((cmd + "\n").encode())
  File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 295, in write
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\USBPrinting\PrinterConnection.py", line 307, in _sendCommand
    self._serial.write(command)
  File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 295, in write
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\USBPrinting\PrinterConnection.py", line 312, in _sendCommand
    self._serial.write((cmd + "\n").encode())
  File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 295, in write
  File "C:\Program Files (x86)\Cura_15.06.03\plugins\USBPrinting\PrinterConnection.py", line 307, in _sendCommand
    self._serial.write(command)
  File "c:\dev\python34\lib\site-packages\serial\serialwin32.py", line 295, in write
serial.serialutil.SerialTimeoutException: Write timeout
```

KJHamblin, 2015-08-10:
Is ```"machine_gcode_flavor": { "default": "RepRap (Marlin/Sprinter)" },``` a valid default option for gcode flavor?  Perhaps Cura is sending commands the printer does not recognize?

nallath, 2015-08-10:
That's not the problem. It is trying to send a command but it is unable to do so. It keeps on trying to do this until the timeout, at which point it gives an error.

KJHamblin, 2015-08-10:
Where am I supposed to see the text from ```Logger.log``` commands (found in ```PrinterConnection.py```)?  I haven't seen anything in Cura that would make me think that it was trying to connect to USB.  Should there have been a pop up window or is there a status bar where this text is displayed?

I haven't even seen the text that should be generated from ```Logger.log("d", "Attempting to connect to %s", self._serial_port)``` in the ```_connect``` function.  

Thanks for the time and help!

nallath, 2015-08-10:
It simply automatically tries all ports upon startup to see what is a printer and what is not. You can find info about the log files in the readme (https://github.com/Ultimaker/Cura#logging-issues)

KJHamblin, 2015-08-10:
I appreciate you pointing me in the right direction.  I'm new to Github and Cura at this level of detail so the help is sincerely appreciated.

But Ha!  Its working now!  I think it was a combination of extra COM ports listed on my computers and my own impatience.  With the current code I would need to wait 45 seconds for it to test the correct baud rate if it started with the correct COM port.  I'd have to wait much longer if there were multiple COM ports on my computer.

That said I have some comments regarding my impatience:

(FYI I created a pull request for everything listed below)

1. Why is ```time.sleep(1.5)``` (Line 210 of ```PrinterConnection.py```) in the ```for baud_rate in self._getBaudrateList():``` loop?  Avoiding communication with the bootloader is a one time thing after powerup.  This is adding an additional 1.5 seconds to every cycle of the loop.

My Suggestion:
Move ```time.sleep(1.5)``` to Line 197 under ```Logger.log("d", "Starting baud rate detection...")``` 

2. Why is the timeout per baud rate tested 15 seconds (```timeout_time = time.time() + 15```)?  This is excessive.  I understand if you were trying to make 10 attempts (15 seconds/time.sleep(1.5)), but that is a lot of down time just sleeping.

My Suggestion:
As previously mentioned, I moved ```time.sleep(1.5)``` out of the loop but I added ```time.sleep(0.1)``` in its place such that Cura was not sending excessive amounts of commands on the USB port.  I also changed ```timeout_time = time.time() + 15``` to ```timeout_time = time.time() + 1```.  A 1 second timeout with a 0.1 second delay still yields 10 attempts.

On another note, I noticed that Cura cannot re-detect a printer if it is disconnected and reconnected without closing and re-opening Cura.  Is this intended?

nallath, 2015-08-11:
The USB printing never got a whole lot of work from us as USB printing is not officially supported for the UM2. It's more of an after thought than a real functionality (this is why it's good that people like you test & improve this!)

Xisle, 2015-08-11:
If I cancel a print, I also have to restart Cura for it to reconnect to my UMO.

KJHamblin, have you tried changing settings in Cura while printing via USB? When I changed "Bottom/Top Pattern" earlier, the printer paused while the mouse button was being pressed. I haven't tried since.

KJHamblin, 2015-08-12:
I have not, but that sounds like an interesting issue so I will give it a go this weekend.

nallath, 2015-08-12:
I'm reworking the USB printing as we speak as we've made some drastic API changes.

closed 2015-08-14 by nallath

Kulitorum, 2016-02-17:
What is the newest version of Cura that can print over USB?

I am debugging a CD-card problem, and I would like to try tethered printing.

Thanks

nallath, 2016-02-17:
The stuff in the 2.1 branch

Kulitorum, 2016-02-17:
Great! - Any compiled version available?

fieldOfView, 2016-02-17:
No, there is no publically available version of 2.1 yet. The latest official release where USB printing is known to be good is probably 15.04.04.