Ultimaker/Cura #233: User submitted profiles not in release package

quillford, 2015-08-10:
Why are user submitted profiles not in the release package? If the user knows how to compile Cura and is willing to, they are going to know how to edit a JSON file to put in basic properties for their printers and there is no point in merging PRs with profiles.

daid, 2015-08-10:
Minor patch releases are done from the release branch, only containing bugfixes:
https://github.com/Ultimaker/Cura/tree/15.06

Next major release will have the extra profiles.

quillford, 2015-08-10:
Thanks for clearing that up. Should people make PRs against 15.06 or master then?

BagelOrb, 2015-08-10:
Against master.

closed 2015-08-10 by BagelOrb

nallath, 2015-08-10:
Unless if it's a bug fix, then you should do it against the version branch (15.06 in this case)