Ultimaker/Cura #234: 15.06.3 - doesn't save info

quillford, 2015-08-10:
Once I restart Cura, it acts as if though it is the first boot and has me select a profile. It doesn't appear to have saved anything from my last session. This is from a "clean" install. I cleared all the old versions of Cura I had previously and installed 15.06.03. I'm on OS X 10.10.4.

nallath, 2015-08-10:
This has probably something to do with the preferences file not being created. Could you check if this is the case?

quillford, 2015-08-10:
The only files it has created are org.pythonmac.unspecified.Cura.plist and org.pythonmac.unspecifid.Cura.savedState.

quillford, 2015-08-10:
I did however notice that it saves a list of recently opened models.

awhiemstra, 2015-08-10:
It should create a directory .cura in your user's home directory. That directory should contain a cura.cfg and a directory settings. In that settings directory should be one or more files describing the machines you added.

daid, 2015-08-10:
@awhiemstra You will "WTF" on this one, but, you cannot always create a .cura in the users home dir. Yes, it breaks POSIX, yes Apple sucks big time. The solution is to store the preferences in:
```
~/Library/Application Support/Cura/
```
See: https://github.com/daid/Cura/blob/SteamEngine/Cura/util/profile.py#L646

quillford, 2015-08-10:
So I left Cura open so that I wouldn't have to recreate my profiles, and I woke up to it showing no profiles or platform in the same instance that I  used to slice a few hours before. When I quit and tried to reopen it gave me

```
Version: 15.06.03
Platform: Darwin-14.4.0-x86_64-i386-64bit
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/Applications/Cura.app/Contents/Resources/cura_app.py", line 9, in <module>
  File "UM/Application.pyc", line 329, in getInstance
  File "cura/CuraApplication.pyc", line 56, in __init__
  File "UM/Qt/QtApplication.pyc", line 76, in __init__
  File "UM/Application.pyc", line 355, in loadMachines
  File "UM/Settings/MachineSettings.pyc", line 87, in loadValuesFromFile
  File "configparser.pyc", line 672, in read
  File "configparser.pyc", line 993, in _read
  File "codecs.pyc", line 319, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 578: invalid start byte
```

quillford, 2015-08-10:
There is a cura.cfg with the following in it

```
[general]
window_height = 931
window_width = 1680
version = 2

[view]

[cura]
active_machine = Prusa i3
active_mode = 1
categories_expanded = ;infill;material;resolution
recent_files = /Users/quillford/Downloads/mega2560_mount.stl;/Users/quillford/Downloads/5mm_Calibration_Steps.stl;/Users/quillford/3D-Modeling/Calibration/small box/box.stl;

[backend]

```

There is also a settings directory with profiles that were modified the same time I was using Cura.

mikenew12, 2015-08-10:
I just spent an hour chasing this down. If you create a custom printer profile, the machine id has to match the name of the json file exactly. So if you create a machine profile with the id of ```prusai3``` and you save it in a file called ```prusa_i3```, Cura will get into this broken state.

The worst part is that deleting and reinstalling the app doesn't help. You have to delete ```~/.cura```.

quillford, 2015-08-10:
@mikenew12 good to know. I added the ability to download the profile in my profile generator because of that. Thanks

mikenew12, 2015-08-11:
@quillford Ha, just realized your'e the one who built the profile generator. Thanks for that :)

nallath, 2015-08-11:
@mikenew12 This was a bug in the code. This should be fixed for 15.10. ID is now used to identify machines, name is now only used as human readable. This means that you can have say 20 profiles that have ID ultimaker2, which gives you a easier way of selecting printer variants.

quillford, 2015-08-11:
@nallath the next major release is scheduled for October?

nallath, 2015-08-11:
Yep

mikenew12, 2015-08-11:
@nallath Maybe there could be a line about that in the readme in the meantime? At the moment it says "change the machine id to something unique"... "put the profile into resources/settings". So if you follow the directions exactly, you wind up with a very broken install of Cura that you can't even fix by reinstalling the app.

Looking forward to 15.10. Despite all the pain from this particular issue, I love the new version of Cura. Great work.

nallath, 2015-08-11:
@mikenew12 Fair point. I'll change it right away.

quillford, 2015-08-12:
I cleared the .cura directory after I installed 15.06.3 and now it saves and loads everything properly.

closed 2015-08-12 by nallath