Ultimaker/Cura #242: Cura 15.06.03 does not work with Ultimaker Original

TheNeuron, 2015-08-11:
Recently I updated Cura to 15.06.03 and updated the firmware of the Ultimaker, now Cura refuses to print since it wants to "Bed heating" even though there is none installed. The printer is defined as an Ultimaker Original and every box concerning the bed heating is disabled. Still in the display of the Ultimaker it says "Bed heating". Since there is no Bed to heat it does not print.

Xisle, 2015-08-11:
Enable "Bed Temperature" and set it to zero.

awhiemstra, 2015-08-11:
The original should have a machine setting that indicates there is no heated bed. The engine currently does not account for that when adding bed temperature. As a workaround, you can set bed temperature to 0 as @Xisle suggests.

Can you open a new bug on the engine for this? (http://github.com/ultimaker/CuraEngine/issues)

TheNeuron, 2015-08-11:
Done

Thanks, the workaround works

closed 2015-08-11 by TheNeuron