Ultimaker/Cura #243: [Discussion] Buildplate meshes

fieldOfView, 2015-08-11:
As more and more printer profiles are added (which is a good thing), the meshes folder is adding a lot of weight. I am not sure if having a buildplate mesh for each possible printer is going to be sustainable. Currently the two rigidbot meshes are more than 30 Mb each, and I know I'll never ever have a use for them. 

Should there be a standard, resizable mesh for most 3d printers?

nallath, 2015-08-11:
Good point. Didn't think about that one before. We could put a limit on the sizes of the plates but that would only decrease the problem so much. I do like the idea that all 3d printers can have their own build plate models in there.

Ghostkeeper, 2015-08-11:
Another limited solution would be to compress them. Most model files compress fairly well, especially ASCII formats. This could even reduce the loading time if the decompressing is not done on the hard disk.

A less limited solution would be to host the model (and printer settings) online somewhere and have Cura download it when the user first selects the printer. But this introduces a dependency on an internet connection, a single-point-of-failure and a centralised point of control, which is something we wouldn't want to happen.

fieldOfView, 2015-08-11:
Some quality control before merging may be in order. Both Rigidbot meshes are way too detailed, the grr_neo does not have much detail at all and, the "hephestos platform" file does not have a platform.

KJHamblin, 2015-08-12:
Ahh... I liked my detail in the Rigidbot meshes!  Admittedly I did not consider file size when creating them which I do see as an issue or the potential to grow into an issue.  Perhaps some guidelines need to be established and documented to help users create these meshes in a sustainable way.

Or maybe, while probably more complicated, is there a way for UM to have a repository of these meshes and setting files and only download them to the user's machine when they select the printer in the wizard?  That would require an internet connection but then the user would only have the files for the printers they will actually use.

I suppose another option for those who are concerned about disk space would be to give the user the option to permanently remove printers from the wizard which would consequently delete the corresponding meshes and .json setting files.  This would still increase Cura's default disk size as the users would need to take action to remove the printers, so this might not be a real solution to the problem.

KJHamblin, 2015-08-12:
And yet another option would be to make the printer selection process part of the installation process such that only the necessary printer files are installed... but the installation executable file size still grows...

awhiemstra, 2015-08-12:
It seems to me the cleanest solution is to just establish some guidelines and maybe additionally provide a default platform mesh that can be used. Removing machine defintions is also something I would be really hesitant to do as it means reinstalling the application if you ever do want to use those definitions.

Downloading things from the internet might be an idea, but then as a separate dialog to manage the available machine definitions. It is also something that will not have priority for a while since we have far more important things to take care of for the moment.

Suggested guidelines:

- Binary STL preferred, use OBJ if you need more features than provided by STL. (Specifically, vertex normals and texture coordinates)
- Files should not be larger than 5 MiB.
- Try and make the platform distinct so you can easily see what machine you are using.

@KJHamblin Looking at the model, I think you can keep a lot of detail while reducing the file size by reducing the smoothness of things like the springs.

KJHamblin, 2015-08-12:
I agree.  I will start messing with that tonight.

fieldOfView, 2015-08-12:
I think 5 Mb is still way too large; that is more than 10 times as large as the Ultimaker meshes.

Should we include a rule that says that the mesh should resemble the actual buildplate of the physical printer? I would also stress that you are in no way obligated to make a mesh; there is no shame in running a printer profile without a buildplate mesh; Cura makes it look nice either way.

BagelOrb, 2015-08-12:
@fieldOfView I don't see a use for that rule. We will not be able to verify it easily for printers we don't own. However I think it is a good rule of thumb for the printer manufacturers introducing these profiles.

Indeed a limit of 2MB should be good enough.

We should make these rules clear. Communicate them to users uploading profiles at a logical place (the wiki?)
Furthermore, we should remove any profile mesh not adhering to the rules, with a clear commit message.

fieldOfView, 2015-08-12:
The "has to resemble the actual buildplate" rule I suggested would hopefully stimulate better quality meshes. For me there is no added value in a "buildplate mesh" like the grr_neo one, which is basically a slab with some text (not even the correct logo of the printer!).

I think 2MB is still very large. 10 printers added would be 20 Mb. And it is not just the weight of the files, but also Cura having to draw more faces than necessary. I had a quick go at editing one of the Rigidbot meshes. Removing the 4 springs and saving it as binary STL made a 550Kb mesh.

KJHamblin, 2015-08-12:
@fieldOfView Thanks for looking at the file.  I have been working on it a bit myself and have it down to 600kb with the spring, but with noticeable flaws and 1.3MB with no noticeable loss of quality.  I removed the spring from the model and used decimate in Blender before resembling the model.  I really wasn't thinking when I put these together but I'll have them fixed and a new pull request in tonight or tomorrow.  Thanks for pointing out my mistake.

May I probe for some more background into your concern on the file sizes and number of faces on these meshes?  Obviously we don't want others to make the mistakes I made and push the Cura distribution size into the GB range, but I was wondering if there is a deeper concern you have.

fieldOfView, 2015-08-12:
> May I probe for some more background into your concern on the file sizes and number of faces on these meshes?

The downloadsize for all of Cura at the moment (with just the Ultimaker buildplate meshes) is 35Mb. 2.6 Mb for two added printer-profiles may not yet seem significant, but 25 printer meshes like that doubles the download size.

But let's not make a big issue out of this. I still say these meshes need quality control before they are merged into the repo, just like code is reviewed and tested before it gets merged. Shitty meshes make Cura look shitty. Cura should not run shitty.

fieldOfView, 2015-08-12:
As a final suggestion, couldn't profiles be something a 3rd party printer-producer or community member supplies for download instead of being in the main distribution? Preferably as a single file (printer profile + mesh + optional texture + some sort of manifest in a zipped file named .curaprofile) that is downloaded as a separate file and installed into the Cura (settings) folder when opened by Cura?

KJHamblin, 2015-08-12:
Ok, that's fair.  I just wanted to get a better idea of your thoughts.

awhiemstra, 2015-08-13:
Thing is, the installers are heavily compressed. I do not know how well binary STL compresses, but for example an OBJ file can be compressed pretty well since it is just a text file. So even if we make the file size limit 5 MiB the installer size will not necessarily increase by the same amount.

awhiemstra, 2015-08-13:
Doing a quick test, with basic settings I can get a binary STL reduced to around 20% of its original size. So a 2MiB file would reduce to ~400KiB.

fieldOfView, 2015-08-13:
This is not an important issue, so this is my last word on the matter (promise).

The old Cura had become unmaintainable mess (not my words), which is why a new Cura was made. What I would hate to see is another mess. With any open source project, curation is key. Keep the standards high. Keep asking if something you want to add to the "core" download really needs to be in the core download (and is of a good enough quality) because it will benefit the majority of users.

awhiemstra, 2015-08-13:
The unmaintainable mess was more from the code side, but you're right, we should be careful. As I said, I would not necessarily mind implmenting some kind of machine manager, but the trouble with that is having the time to actually implement it. :)

ErikDeBruijn, 2015-08-14:
Apart from package size and loading speed, a large mesh might also mean worse performance orbiting around simple models. You'll have a GPU load even though you haven't even loaded any high-polygon models. I think attention to this will be helpful.
If we do want to reduce storage even further, for mesh compression there's the .ctm format. OpenCTM makes models absolutely tiny. It also supports textures.

awhiemstra, 2015-08-17:
Since it is one batch of static geometry, I am not too worried about performance. You would need a platform mesh of more than a million triangles before there would be any noticable impact on a recent Intel chip, let alone a dedicated AMD or NVidia card. That said, setting an upper limit for the amount of vertices does seem like a good idea.

innovo3d, 2015-09-08:
Hi everyone.  If I can put in my 2 cents, first and foremost I agree with keeping the file size as low as possible, 500kb max sounds reasonable. However, it may not be such a terrible idea to revisit the idea of having the installer download the additional files. Cura requires an internet connection to be downloaded anyway. Sure there are places in the world where internet connection may not be as stable or reliable, but I'm sure in these cases people will download Cura with the right settings in advance. 

We could have an installer/downloaded (lessnthan 1mb file) that opens a dialogue to select the printer settings before downloading Cura with selected settings and platform models. If you aren't connected it could simply just load the Ultimaker or a default platform.

closed 2017-01-11 by fieldOfView