Ultimaker/Cura #244: [Request] Visual marker for origin

quillford, 2015-08-12:


Ghostkeeper, 2015-08-12:
The origin is always in the centre. Perhaps it is useful, but I think the effect should be fairly subtle. Perhaps it is good style to draw a small cross of like 10 pixels wide in the centre in a slightly darker shade of grey than the platform surface. Is that the sort of thing that you meant?

quillford, 2015-08-12:
I mean the origin of the printer (i.e. 0,0).

quillford, 2015-08-12:
It was like this in daid/Cura 
[image]

BagelOrb, 2015-08-12:
Why is that origin of any importance to the user?

quillford, 2015-08-12:
If there is no platform model for their printer's profile, it shows the orientation of the model on the bed.

BagelOrb, 2015-08-12:
Every printers profile should have a platform model. We're not going to introduce features which only have use when content supplied by other printers is lacking.

closed 2015-08-12 by BagelOrb