Ultimaker/Cura #246: An uncaught exception has occurred!

Thebard108, 2015-08-12:
Just updated to version: 15.06.03 and the program now refuses to run.
it was working this morning when i was running version 15.06.02
i've tried restarting the computer & re installing the program... same error pops up.

Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 124, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__
KeyError: 'general'

i'm now going to download the old version and see if that works

nallath, 2015-08-13:
Delete your preferences file. This is an issue that occurs when old versions were installed.

closed 2015-08-13 by nallath

Thebard108, 2015-08-13:
Where is the preferences file your refering to located, since it's not the one under C:\Program Files\Cura_15.06.03\qml\UM\Preferences, as this is deleted when uninstalling the program?

nallath, 2015-08-13:
It should be a file with a .cfg extention. Not sure where it is exactly on windows.

awhiemstra, 2015-08-13:
See https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

Thebard108, 2015-08-13:
Thank you, its working now :)