Ultimaker/Cura #248: 15.06.03 "Errors in 'Cura.exe' - See the logfile ...

Fuzionist1, 2015-08-12:
Nearly (~90%) every time I exit Cura, a popup displays
[image]

The logfile includes the following:

C:\Program Files (x86)\Cura_15.06.03\library.zip\UM\Mesh\MeshData.py:422: RuntimeWarning: invalid value encountered in true_divide
C:\Program Files (x86)\Cura_15.06.03\library.zip\UM\Mesh\MeshData.py:423: RuntimeWarning: invalid value encountered in true_divide
C:\Program Files (x86)\Cura_15.06.03\library.zip\UM\Mesh\MeshData.py:424: RuntimeWarning: invalid value encountered in true_divide
Traceback (most recent call last):
  File "c:\dev\uranium\UM\Qt\Bindings\ControllerProxy.py", line 26, in setActiveView
  File "c:\dev\uranium\UM\Controller.py", line 85, in setActiveView
  File "c:\dev\uranium\UM\Signal.py", line 94, in emit
  File "c:\dev\python34\lib\weakref.py", line 387, in items
RuntimeError: dictionary changed size during iteration

I run Cura in Administrator mode.

nallath, 2015-08-13:
I've had these a few times as-well. Thanks for reporting it, now we know it's not just linux that has this issue.

Fuzionist1, 2015-08-13:
I should mention I'm windows 7

~Nick

pascalbuijs, 2015-09-02:
Same errors here with Windows 10 and Windows 8.1. Running in administrator mode
The print also appears to fail when i close cura and gives the error.

RuntimeError: dictionary changed size during iteration
C:\Program Files (x86)\Cura_15.06.03\library.zip\UM\Mesh\MeshData.py:422: RuntimeWarning: invalid value encountered in true_divide
C:\Program Files (x86)\Cura_15.06.03\library.zip\UM\Mesh\MeshData.py:423: RuntimeWarning: invalid value encountered in true_divide
C:\Program Files (x86)\Cura_15.06.03\library.zip\UM\Mesh\MeshData.py:424: RuntimeWarning: invalid value encountered in true_divide

closed 2015-09-15 by nallath