Ultimaker/Cura #250: Cannot save gcode

BagelOrb, 2015-08-13:
After slicing the "Save to [Disk/SD]" button stays gray.

Expected : let me save the gcode, please!
Actual : (
Steps to reproduce : load file

======================
Version: master
Platform: Linux-3.19.0-25-generic-x86_64-with-Ubuntu-15.04-vivid
Qt: 5.4.1
PyQt: 5.4.1
======================
log:
2015-08-13 12:20:39,071 - INFO - Loaded plugin FileLogger
2015-08-13 12:20:39,074 - INFO - Loaded plugin GCodeWriter
2015-08-13 12:20:39,076 - WARNING - Could not determine application version from string master, not checking for updates
2015-08-13 12:20:39,077 - INFO - Loaded plugin LayerView
2015-08-13 12:20:39,081 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-08-13 12:20:39,083 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-13 12:20:39,083 - ERROR - Could not find plugin 3MFReader
2015-08-13 12:20:39,085 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-13 12:20:39,093 - INFO - Loaded plugin USBPrinting
2015-08-13 12:20:39,209 - INFO - Loaded plugin CuraEngineBackend
2015-08-13 12:20:39,287 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-08-13 12:20:39,287 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-08-13 12:20:39,288 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-08-13 12:20:39,288 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
2015-08-13 12:20:39,568 - ERROR - Error when loading file fdmprinter.json.orig: Expecting ',' delimiter: line 102 column 1 (char 2462)
2015-08-13 12:20:39,759 - DEBUG - Loaded theme /home/tim/Development/Cura/resources/themes/cura
2015-08-13 12:20:41,623 - INFO - Started engine process: /home/tim/Development/CuraEngine/build/CuraEngine
2015-08-13 12:20:41,631 - DEBUG - Backend connected on port 49678
2015-08-13 12:20:43,958 - DEBUG - Loaded a mesh with 36 vertices

===================================

Engine log:
Calling engine with: ['/home/tim/Development/CuraEngine/build/CuraEngine', 'connect', '127.0.0.1:49678', '-j', '/home/tim/Development/Cura/cura/../resources/settings/fdmprinter.json', '-vv']

[... ommitted stuff...]

End sliced object called. sliced objects 1 object count: 1
Total time elapsed  7.40s.

nallath, 2015-08-21:
Is this still an issue? You also have an error with fdmprinter.json.orig, which could cause this.

BagelOrb, 2015-08-24:
It's working now. 
Yay!

closed 2015-08-24 by BagelOrb

s4mdf0o1, 2015-12-07:
hi, what have you for this to work, while I'm in the same situation.
What I can say is that the lonely save is grayed the same,
File>Print: Says "saved to SDCard", but I've got an empty gcode file...
the logs : "Unknown exception"
I think it could be a design error of the 3D object,
because I've drowned one to test sizes and calibration with Blender and start/end-gcode

BagelOrb, 2015-12-07:
@s4mdf0o1 Eventhough a model has faults, still some gcode should be produced.
Can you maybe provide me with the faulty model?

Then we can see what went wrong.

PS: this is a separate issue from the original issue in this thread.

s4mdf0o1, 2015-12-07:
it's just a cube. 2cmx1cmx0.12mm
-right testing my printer's parameters :)
I realised that 0.12mm heigth, made the 'Save' button get gray
Then tested with 0.18, all went good...
That's why I'm talking about faulty 3D image :
I've seen many of topics saying "save button gray", then later "all is good", without response.
So I'm trying to give one :) ;)

BagelOrb, 2015-12-07:
A cube with height 0.12mm has no layers, because the default setting for Initial Layer Height is 0.3mm, so the first layer is sliced at 0.15mm and printed at 0.3.

If I remember correctly this bug has been solved in the meantime.