Ultimaker/Cura #253: Sorry folks, new to Cura building

ze897c, 2015-08-14:
This is probably my fault, but the pop-up asked me to post.

Version: master
Platform: Darwin-14.5.0-x86_64-i386-64bit
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/usr/local/Cura.app/Contents/Resources/__boot__.py", line 77, in <module>
  File "/usr/local/Cura.app/Contents/Resources/__boot__.py", line 62, in _run
  File "/usr/local/Cura.app/Contents/Resources/cura_app.py", line 16, in <module>
  File "UM/Application.pyc", line 301, in getInstance
  File "cura/CuraApplication.pyc", line 58, in __init__
  File "UM/Qt/QtApplication.pyc", line 78, in __init__
  File "cura/CuraApplication.pyc", line 109, in _loadPlugins
RuntimeError: Could not load the backend plugin!

nallath, 2015-08-14:
Could you also post the log file? Might be that there is some more info there.

awhiemstra, 2015-08-14:
The backend plugin fails to load. In the log file, there will be a reason why. Most likely you failed to build/install either libArcus or Protobuf.

ErikDeBruijn, 2015-08-14:
Hi, I'm getting the same error after @awhiemstra had commited a change to libArcus with me.
```
Version: master
Platform: Darwin-14.4.0-x86_64-i386-64bit
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/private/tmp/cura-build/build/dist/Cura.app/Contents/Resources/__boot__.py", line 77, in <module>
  File "/private/tmp/cura-build/build/dist/Cura.app/Contents/Resources/__boot__.py", line 62, in _run
  File "/private/tmp/cura-build/build/dist/Cura.app/Contents/Resources/cura_app.py", line 16, in <module>
  File "UM/Application.pyc", line 301, in getInstance
  File "cura/CuraApplication.pyc", line 58, in __init__
  File "UM/Qt/QtApplication.pyc", line 78, in __init__
  File "cura/CuraApplication.pyc", line 109, in _loadPlugins
RuntimeError: Could not load the backend plugin!
```

The cura.log gives me:
```
2015-08-14 17:25:04,187 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-14 17:25:04,187 - ERROR - Could not find plugin 3MFReader
2015-08-14 17:25:04,190 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-14 17:25:04,194 - ERROR - Import error loading module CuraEngineBackend: No module named 'cura.OneAtATimeIterator'
2015-08-14 17:25:04,194 - ERROR - Could not find plugin CuraEngineBackend
2015-08-14 17:25:04,197 - ERROR - Import error loading module CuraEngineBackend: No module named 'cura.OneAtATimeIterator'
2015-08-14 17:25:04,201 - INFO - Loaded plugin GCodeWriter
2015-08-14 17:25:04,205 - INFO - Loaded plugin LayerView
 ```

I removed all plugins I had installed manually to ensure it's not related to that.

ze897c, 2015-08-14:
The only log file I currently see in **build** is Protobuf.log. Nothing in CuraEngine-prefix/tmp or Cura-prefix/tmp. Where else shall I look?

```
<<Protobuf.log>>

/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__init__.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/api_implementation.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/containers.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/decoder.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/encoder.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/enum_type_wrapper.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/message_listener.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/python_message.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/type_checkers.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/wire_format.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/__init__.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/descriptor.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/descriptor_pb2.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/compiler/__init__.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/compiler/plugin_pb2.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/message.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/descriptor_database.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/descriptor_pool.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/message_factory.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/proto_builder.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/pyext/__init__.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/pyext/cpp_message.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/reflection.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/service.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/service_reflection.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/symbol_database.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/text_encoding.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/text_format.py
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/__init__.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/api_implementation.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/containers.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/decoder.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/encoder.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/enum_type_wrapper.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/message_listener.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/python_message.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/type_checkers.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/internal/__pycache__/wire_format.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/__pycache__/__init__.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/__pycache__/descriptor.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/__pycache__/descriptor_pb2.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/compiler/__pycache__/__init__.cpython-34.pyc
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/google/protobuf/compiler/__pycache__/plugin_pb2.cpython-34.pyc
[... cut ...]
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/protobuf-3.0.0a3.dev0-py3.4.egg-info/requires.txt
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/protobuf-3.0.0a3.dev0-py3.4.egg-info/SOURCES.txt
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/protobuf-3.0.0a3.dev0-py3.4.egg-info/top_level.txt
/Users/mattpatterson/devel/cura-build/build/inst/lib/python3.4/site-packages/protobuf-3.0.0a3.dev0-py3.4-nspkg.pth
 
'''

ErikDeBruijn, 2015-08-14:
@awhiemstra and I looked into the build and he updated several repositories. If you run cura-build again, we got it working.
We also cleaned out the files in `~/.cura/settings/`. If it goes further but fails on the confguration, you could try the same thing.
Remove the `dist/Cura.app` and re-run `make`.
Note that we're using https://github.com/Ultimaker/cura-build on OS X.

B.t.w. I used `~/.cura/cura.log` to produce the output. It updates when I start the newly built Cura.app.

ze897c, 2015-08-14:
Sorry for being dense again, but pull on cure-build master only brings one modified file (setup_osx.py.in): the cure-build gitpository is meant to be monolithic, yes?

```
<<~/.cura/cura.log>>
2015-08-13 17:33:00,472 - INFO - Loaded plugin FileLogger
2015-08-13 17:33:00,477 - INFO - Loaded plugin LocalFileOutputDevice
2015-08-13 17:33:00,481 - INFO - Loaded plugin CameraTool
2015-08-13 17:33:00,493 - INFO - Loaded plugin MirrorTool
2015-08-13 17:33:00,510 - INFO - Loaded plugin RotateTool
2015-08-13 17:33:00,517 - INFO - Loaded plugin ScaleTool
2015-08-13 17:33:00,522 - INFO - Loaded plugin SelectionTool
2015-08-13 17:33:00,529 - INFO - Loaded plugin TranslateTool
2015-08-13 17:33:00,577 - INFO - Checking for new version of cura
2015-08-13 17:33:00,578 - INFO - Loaded plugin UpdateChecker
2015-08-13 17:33:00,591 - INFO - Loaded plugin MeshView
2015-08-13 17:33:00,594 - INFO - Loaded plugin WireframeView
2015-08-13 17:33:00,598 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-13 17:33:00,598 - ERROR - Could not find plugin 3MFReader
2015-08-13 17:33:00,602 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-13 17:33:00,607 - ERROR - Import error loading module CuraEngineBackend: No module named 'Arcus'
2015-08-13 17:33:00,607 - ERROR - Could not find plugin CuraEngineBackend
2015-08-13 17:33:00,611 - ERROR - Import error loading module CuraEngineBackend: No module named 'Arcus'
2015-08-13 17:33:00,615 - INFO - Loaded plugin GCodeWriter
2015-08-13 17:33:00,625 - INFO - Loaded plugin LayerView
2015-08-13 17:33:00,642 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-08-13 17:33:00,680 - INFO - Loaded plugin USBPrinting
2015-08-13 17:33:01,525 - WARNING - Could not determine application version from string master, not checking for updates
2015-08-13 17:43:31,499 - INFO - Loaded plugin FileLogger
2015-08-13 17:43:31,505 - INFO - Loaded plugin LocalFileOutputDevice
2015-08-13 17:43:31,511 - INFO - Loaded plugin CameraTool
2015-08-13 17:43:31,519 - INFO - Loaded plugin MirrorTool
2015-08-13 17:43:31,537 - INFO - Loaded plugin RotateTool
2015-08-13 17:43:31,545 - INFO - Loaded plugin ScaleTool
2015-08-13 17:43:31,551 - INFO - Loaded plugin SelectionTool
2015-08-13 17:43:31,558 - INFO - Loaded plugin TranslateTool
2015-08-13 17:43:31,603 - INFO - Checking for new version of cura
2015-08-13 17:43:31,604 - INFO - Loaded plugin UpdateChecker
2015-08-13 17:43:31,617 - INFO - Loaded plugin MeshView
2015-08-13 17:43:31,622 - INFO - Loaded plugin WireframeView
2015-08-13 17:43:31,626 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-13 17:43:31,626 - ERROR - Could not find plugin 3MFReader
2015-08-13 17:43:31,630 - ERROR - Import error loading module 3MFReader: No module named 'untangle'
2015-08-13 17:43:31,636 - ERROR - Import error loading module CuraEngineBackend: No module named 'Arcus'
2015-08-13 17:43:31,636 - ERROR - Could not find plugin CuraEngineBackend
2015-08-13 17:43:31,641 - ERROR - Import error loading module CuraEngineBackend: No module named 'Arcus'
2015-08-13 17:43:31,647 - INFO - Loaded plugin GCodeWriter
2015-08-13 17:43:31,656 - INFO - Loaded plugin LayerView
2015-08-13 17:43:31,670 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-08-13 17:43:31,703 - INFO - Loaded plugin USBPrinting
2015-08-13 17:43:32,017 - WARNING - Could not determine application version from string master, not checking for updates

awhiemstra, 2015-08-17:
You will need to remove the Arcus-prefix directory. We fixed the install location of Arcus' files on Mac OSX but the location is cached so is not changed when the underlying CMake file changes.

closed 2015-08-27 by nallath