Ultimaker/Cura #254: Unexpected Error at the End of Successful Print - USB Related

KJHamblin, 2015-08-14:
Application Version: 15.06.03
Platform: Windows 7
Qt: Unsure
PyQt: Unsure
Display Driver: Unsure

Steps to Reproduce:
1. Power on Arduino Mega 2560 based printer 
2. Connect USB to computer
3. Open Cura 15.06.XX
4. Load .STL file
5. Print the file 
6. Wait for the print to complete

Expected results:
Print finishes correctly, USB Dialog box closes or says something like "Complete!"

Actual Results:
printer finishes its routine as normal including moving the extruder up and away from the print, homing the Y and then moving the completed part forward.  BUT the USB Dialog box stay open, states an unexpected error occurred, which is a Serial Time Out error.

Additional Information:
Since @nallath is reworking the USB I wanted to mention this as well.  This has happened every time I have completed a print.  It doesn't bother me too much since the print is successful, but its still worth mentioning.

nallath, 2015-08-14:
You will be able to check the changes in the 15.10 release.

KJHamblin, 2015-08-14:
Sounds good; I look forward to it!

BagelOrb, 2017-07-21:
Probably doesn't happen any more.

closed 2017-07-21 by BagelOrb