Ultimaker/Cura #263: Fan speed tooltip

ErikDeBruijn, 2015-08-17:
Not sure, but could it be that minimum and maximum were swapped in this explanation?

[image]

I'm a big "fan" of tooltips, however :)

Xisle, 2015-08-17:
While on the topic of tooltips..

The tooltips for Machine Settings have an issue with the display timer. It looks like the mouseEntered() is being handled before mouseExited() and mouseEntered() isn't stopping the previous Timer.

Xisle, 2015-08-17:
https://www.youtube.com/watch?v=TJU-aoBvDw0&feature=youtu.be

BagelOrb, 2015-08-17:
@Xisle please submit a new issue. These two are unrelated.

@ErikDeBruijn the description is correct, though hard to understand. A later cannot take less time than the minimal layer time, at which points the fan will be at Max speed. For  layers taking longer the fan speed reduces, until a layer takes longer than "minimal layer time minimal fan speed".

I should maybe rephrase the tooltip...

Xisle, 2015-08-17:
I was already working on it ;)

closed 2015-08-26 by nallath