Ultimaker/Cura #264: Machine Settings Tooltip Timer

Xisle, 2015-08-17:
Cura 15.06.03
OS X 10.9.5

The tooltips for Machine Settings have an issue with the display timer. It looks like the mouseEntered() is being handled before mouseExited() and mouseEntered() isn't stopping the previous Timer.

https://www.youtube.com/watch?v=TJU-aoBvDw0&feature=youtu.be

Xisle, 2015-08-17:
With the length of some tooltips, maybe no Timer would be better?

nallath, 2015-08-18:
No timer also feels a bit off. I will experiment a bit with shorter timers, see how that works. It doesn't have that high of a priority though.

Xisle, 2015-09-18:
Shortening the timer would likely not leave enough time to read the longer tooltips. No timer seems more standard; display tooltip until mouseExited. Not everyone reads at the same rate.

I just tried 15.09.82 and the tooltips are now gone in Cura > Preferences > Settings. They work great in the Print Setup section though.

Ghostkeeper, 2015-09-18:
I agree with Xisle. Having it the way that users are most used to from web browsers is best, I think.

awhiemstra, 2015-09-25:
15.10 uses a different mechanism for tooltips that is closer to the platform implementation. The timer issue should now be resolved.

closed 2015-09-25 by awhiemstra