Ultimaker/Cura #265: Cura crashes when launched

richdaddy89, 2015-08-17:
Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 76, in __init__
  File "c:\dev\uranium\UM\Application.py", line 355, in loadMachines
  File "c:\dev\uranium\UM\Settings\MachineSettings.py", line 87, in loadValuesFromFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1044, in _read
configparser.DuplicateSectionError: While reading from 'C:\\Users\\Elite/AppData/Local/cura\\settings\\Ultimaker+2.cfg' [line 39]: section 'blackmagic' already exists

Xisle, 2015-08-18:
It looks like you have two [blackmagic] sections in your Ultimaker+2.cfg file. Delete one.

richdaddy89, 2015-08-18:
That worked thanks!

closed 2015-08-18 by nallath