Ultimaker/Cura #266: Error: trying to write progress

BagelOrb, 2015-08-18:
I'm getting an error on the command line and Cura is not slicing.

Version: master
Platform: Linux-3.19.0-25-generic-x86_64-with-Ubuntu-15.04-vivid
Qt: 5.4.1
PyQt: 5.4.1

log:
-----
Development/Cura$ PYTHONPATH=../libArcus/python/:../Uranium/. python3 cura_app.py 
[ omitted]
... 
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/ScrollView.qml:202: TypeError: Cannot assign to read-only property "progress"
2015-08-18 13:31:28,847 - DEBUG - Loaded theme /home/tim/Development/Cura/resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null

nallath, 2015-08-18:
Those errors are QML only, so they should not affect the slicing.

BagelOrb, 2015-08-18:
So should I file a new report for the not slicing bug?

closed 2015-09-15 by nallath