Ultimaker/Cura #267: Feature: drop on buildplate / level z

BagelOrb, 2015-08-18:
When loading a model part of it may stick out under the build platform.

Steps: Load a model by dragging, or transform a mesh
Expected: mesh *sits* on buildplate
Actual: mesh cut halfway through buildplate

Solution:
- either a richt click context menu item to level the z position
- or automatically level z position when loaded and on some transformations (scale, rotate x or y)

[image]

nallath, 2015-08-18:
This is by design. Moving the model through the build plate lets you cut off a certain distance at the bottom.

BagelOrb, 2015-08-18:
Ok.

However, this is not what a typical user expects, given that this functionality wasn't default in all past Cura versions. I've found several STLs I had laying around which are sunk into the built plate very slightly; without paying too much attention I would surely have overlooked the fact that they were exported with the Z coordinates at the wrong place.

Though I think moving the object through the build plate provides valuable functionality I think that by default when loading a mesh the origin of the coordinate system should be corrected.

My issue still stands; when **loading** a mesh it shouldn't cut off a certain distance at the bottom.

nallath, 2015-08-18:
That wasn't quite clear in the original message, but fair enough. When loading they should be centred correctly (completely above the platform). For all the objects that I tested this was the case. Could you send me the objects where this is not the case?

closed 2015-08-20 by nallath