Ultimaker/Cura #268: Mesh color when normal is aligned with camera

BagelOrb, 2015-08-18:
When looking straight at a face of a model, the color is too bright.

[image]

BagelOrb, 2015-08-24:
current master:
[image]

BagelOrb, 2015-08-24:
[image]

nallath, 2015-08-24:
Lowering the specular even further will pretty much remove the specular effect. This only happens in this very specific orientation.

closed 2015-08-24 by nallath

BagelOrb, 2015-08-24:
Working
[image]