Ultimaker/Cura #272: reporting bug:uncaught exception

zljones, 2015-08-18:
Version: 15.06.03
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1088, in _read
configparser.ParsingError: Source contains parsing errors: 'C:\\Users\\KarenWarner/AppData/Local/cura\\cura.cfg'
	[line 15]: '0150706-25888-a9thn5-0.stl;\n'

Ghostkeeper, 2015-08-18:
Thanks for the report!

Though the traceback helps, I can't reproduce this exception with just this. What were you doing that caused the exception? And perhaps you could share the contents of that file, "C:\Users\KarenWarner\AppData\Local\cura\cura.cfg"? That would probably help getting this fixed.

nallath, 2015-08-19:
If you want a quick workarround; Deleting it and re-running cura should fix it.

zljones, 2015-08-19:
hi thanks, I had been using the most recent version of Cura, then it refused to open, deleted installation, re installed. also tried on another computer same thing happened.just got the obove exception/bug. hav now re installed an older cura version its working fine.
thanks

nallath, 2015-08-26:
I'm not able to reproduce this issue in any way. I've added more logging for the 15.10 version, so i'm hoping that these issues (as we've had multiple of these) can be fixed properly.

closed 2015-08-26 by nallath