Ultimaker/Cura #274: Crash Upon Launch, relating to lists

MiniguyBrendan, 2015-08-19:
Version: master (github, probably closest to 15.06.03)
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "C:\Users\Owner\Cura\UM\Qt\Bindings\Models.py", line 114, in availableMachinesModel
    self._available_machines_model = AvailableMachinesModel.AvailableMachinesModel()
  File "C:\Users\Owner\Cura\UM\Qt\Bindings\AvailableMachinesModel.py", line 34, in __init__
    self._updateModel()
  File "C:\Users\Owner\Cura\UM\Qt\Bindings\AvailableMachinesModel.py", line 120, in _updateModel
    for item in sorted(_machines_by_ultimaker):
TypeError: unorderable types: ListModel() < ListModel()

nallath, 2015-08-20:
Master is very very far from 15.06.03 ;)

MiniguyBrendan, 2015-08-20:
Oh, sorry. Didn't realize. Still, does anyone have a fix? If not, does anyone have a build of this? I need the reworked USB API.

nallath, 2015-08-21:
It's the first time i've seen the issue. I'll have a look and see if I can fix it.

nallath, 2015-08-21:
I've added some code that I hope fixes the problem. Could you check if this is the case? The change is in Uranium, so you need to pull code from there.

MiniguyBrendan, 2015-08-21:
Version: master
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 17, in <module>
    app.run()
  File "C:\Users\Owner\Cura\cura\CuraApplication.py", line 163, in run
    self.setActiveMachine(machine)
  File "C:\Users\Owner\Cura\UM\Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "C:\Users\Owner\Cura\UM\Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "C:\Users\Owner\Cura\UM\Scene\Platform.py", line 49, in _onActiveMachineChanged
    self.setMeshData(app.getMeshFileHandler().read(Resources.getPath(Resources.MeshesLocation, mesh), center = False).getMeshData())
AttributeError: 'NoneType' object has no attribute 'getMeshData'

You fixed the first error, thanks!!! But there's another one now, unfortunately. 

When I run as admin:

Version: master
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 17, in <module>
    app.run()
  File "C:\Users\Owner\Cura\cura\CuraApplication.py", line 163, in run
    self.setActiveMachine(machine)
  File "C:\Users\Owner\Cura\UM\Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "C:\Users\Owner\Cura\UM\Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "C:\Users\Owner\Cura\cura\CuraApplication.py", line 468, in _onActiveMachineChanged
    self._volume.rebuild()
  File "C:\Users\Owner\Cura\cura\BuildVolume.py", line 78, in rebuild
    minW = -self._width / 2
TypeError: bad operand type for unary -: 'NoneType'

When I run as regular user: 

Version: master
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 17, in <module>
    app.run()
  File "C:\Users\Owner\Cura\cura\CuraApplication.py", line 163, in run
    self.setActiveMachine(machine)
  File "C:\Users\Owner\Cura\UM\Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "C:\Users\Owner\Cura\UM\Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "C:\Users\Owner\Cura\cura\PrintInformation.py", line 188, in _onActiveMachineChanged
    self._updateTimeQualitySettings()
  File "C:\Users\Owner\Cura\cura\PrintInformation.py", line 198, in _updateTimeQualitySettings
    self._low_quality_settings.loadSettingsFromFile(Resources.getPath(Resources.SettingsLocation, self._current_settings.getTypeID() + ".json"))
  File "C:\Users\Owner\Cura\UM\Resources.py", line 51, in getPath
    raise FileNotFoundError("Could not find resource {0} in {1}".format(args, type))
FileNotFoundError: Could not find resource ('unknown.json',) in 2

Thanks in advance!

nallath, 2015-08-24:
It seems to be unable to find the location of the machine. Why are you even trying to use the master if you need our help fixing those errors? Running it from code is an advanced way of doing it, intended for software engineers who want to contribute.

bijanbina, 2015-08-27:
I Have the same error
Version: master
Platform: Linux-3.14.51-1-lts-x86_64-with-arch
Qt: 5.5.0
PyQt: 5.5

Exception:
Traceback (most recent call last):
  File "/usr/bin/cura_app.py", line 17, in <module>
    app.run()
  File "/usr/lib/python3.4/site-packages/cura/CuraApplication.py", line 168, in run
    self.setActiveMachine(machine)
  File "/usr/lib/python3.4/site-packages/UM/Application.py", line 261, in setActiveMachine
    self.activeMachineChanged.emit()
  File "/usr/lib/python3.4/site-packages/UM/Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "/usr/lib/python3.4/site-packages/UM/Scene/Platform.py", line 49, in _onActiveMachineChanged
    self.setMeshData(app.getMeshFileHandler().read(Resources.getPath(Resources.MeshesLocation, mesh), center = False).getMeshData())
AttributeError: 'NoneType' object has no attribute 'getMeshData'

MiniguyBrendan, 2015-08-27:
Well, when is the next time you are planning on doing a release? Cura doesn't work with my printer atm because of the old USB code. The GitHub Cura has an updated version from what I can tell, and I'm hoping it will fix the problem.

bijanbina, 2015-08-27:
I Use some dirty hack to fix this change UM/Scene/Platform.py _onActiveMachineChanged function as this
```
    def _onActiveMachineChanged(self):
        if self._settings:
            self.setMeshData(None)

        app = Application.getInstance()
        self._settings = app.getActiveMachine()
        if self._settings:
            mesh = self._settings.getPlatformMesh()
            if not mesh:
                mesh = "" # Dirty hack to ensure that if no mesh is set it doesn't crash.
            else:
                self.setMeshData(app.getMeshFileHandler().read(Resources.getPath(Resources.MeshesLocation, mesh), center = False).getMeshData())
                self._texture = self._settings.getPlatformTexture()

                if self._material and self._texture:
                	self._material.setUniformTexture("u_texture", Resources.getPath(Resources.ImagesLocation, self._texture))
```
after that another error pops up
just change following code in /usr/lib/python3.4/site-packages/cura/BuildVolume.py
```
if self._width == 0 or self._height == 0 or self._depth == 0:
```
to 
```
if self._width == 0 or self._height == 0 or self._depth == 0 or self._width is None:
```
Sorry to interrupt i know it's not a good place for this but maybe someone have to build latest version for his/her work

nallath, 2015-08-27:
We will release the 15.10 beta on 1 September.

I don't think it's that dirty of a hack, considering that you document it in your code. If you would add some logging if this occurs (eg; "Unable to get platform mesh") and handle the rest of the error gracefully, you'd have a pretty good catch for this.

If you want you can make a pull request for those changes.

closed 2015-09-15 by nallath