Ultimaker/Cura #275: Cura crashing on opening

cdubs7, 2015-08-21:
Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 124, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 937, in __getitem__
KeyError: 'general'

nallath, 2015-08-21:
Delete the *.cfg file. This issue is fixed for the next release.  You can find the location on https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

closed 2015-08-21 by nallath

cdubs7, 2015-08-21:
thanks for your help, where do I find the *.cfg file? 

Many thanks, 

Clara 

Date: Fri, 21 Aug 2015 00:39:37 -0700

nallath, 2015-08-21:
See the link i posted.

cdubs7, 2015-08-21:
my bad, thanks!